raspberrypi/libcamera #239: build.ninja is not found when attempting to build

Pdom02, 2025-02-11:
This is using a raspberry pi 5 and raspbian OS

I have followed the guide and run the following commands to install the dependencies:

```
sudo apt install -y libboost-dev
sudo apt install -y libgnutls28-dev openssl libtiff-dev pybind11-dev
sudo apt install -y qtbase5-dev libqt5core5a libqt5widgets
sudo apt install -y meson cmake
sudo apt install -y python3-yaml python3-ply
sudo apt install -y libglib2.0-dev libgstreamer-plugins-base1.0-dev

```

It said it couldn't find libqt5widgets but everything else installed.

After this I then ran the normal build commands:

```
git clone https://github.com/raspberrypi/libcamera.git
cd libcamera
meson setup build --buildtype=release -Dpipelines=rpi/vc4,rpi/pisp -Dipas=rpi/vc4,rpi/pisp -Dv4l2=true -Dgstreamer=enabled -Dtest=false -Dlc-compliance=disabled -Dcam=disabled -Dqcam=disabled -Ddocumentation=disabled -Dpycamera=enabled
ninja -C build install
```

Everything ran fine except for ```ninja -C build install```

When running that command I got the following output:
```
ninja: Entering directory `build'
ninja: error: loading 'build.ninja': No such file or directory
```

I've double-checked that I've installed ninja-build and meson so I'm not sure what the issue is, please let me know if you need any clarifying information.

naushir, 2025-02-12:
There are instructions at the end of the document related to this:
https://github.com/raspberrypi/libcamera#troubleshooting

Let me know if you are still unable to resolve this.

kbingham, 2025-02-12:
The error message is not saying the ninja tool is not available, but that it can't find a build.ninja file under the build directory. 

This means your meson setup stage has likely not completed successfully? 

Please check that command and it's output, perhaps posting the full output on the issue to see what it reports

naushir, 2025-02-12:
I've seen this error when there's a conflict between system installs of meson/ninja vs user installs.

Pdom02, 2025-02-13:
Hey thanks for the responses! I'm sorry about not reading more into the documentation before asking this question. Turns out, the issue was with the meson setup step. My computer had an old installation of python and meson was using that and was unable to finish the setup correctly. Thanks again!

closed 2025-02-13 by Pdom02