raspberrypi/libcamera #241: gstreamer : libcamerasrc --info-text %focus value - how to access?

Kletternaut, 2025-02-18:
Hi,
I would like to ask if there is a way for a HQCamera module with rpi5 with current software versions, in conjunction with Hailo8 AI processor, to detect and track an output of --info-text %focus in the gstreamer libcamerasrc. 

I use it like
```
if self.source_type == "rpi
                source_element = f "libcamerasrc name=\\"camera {selected_index}\\" camera-name=\\"{selected_camera_path}\\" ! "

```and it works really well. The only thing I'm missing is access to the --info-text %focus value during runtime.

I would really need this for my little camera project to control a motorised zoom telephoto lens, i.e. the 'automatic' focus function I have implemented.

![Image](https://github.com/user-attachments/assets/f56dc022-4bb5-4798-ab5f-d68217bbaf35)

kbingham, 2025-02-20:
The FoM value should already be being reported as metadata (https://libcamera.org/api-html/namespacelibcamera_1_1controls.html#a3c5f034588ba57aaeb4d0d79ce0735e2) but it's how to get it from gstreamer that's the tough part, and I don't know that one. It looks like you might be using python gstreamer bindings perhaps?

Equally - if you then use that value in your gstreamer components, to decide your focus  you can then set a Manual focus mode (https://libcamera.org/api-html/namespacelibcamera_1_1controls.html#a4822d64036849aca4e912c50a0e608aa) and then you can set the LensPosition (https://libcamera.org/api-html/namespacelibcamera_1_1controls.html#acfb623408035e51db904a0f3928a863e)

So libcamera should already have what you need - but I don't know how you trigger that from gstreamer elements.

6by9, 2025-02-20:
There's nothing stopping lens control being added to IMX477, just that there have been no modules configured to do so (it looks like Arducam do make one - https://www.uctronics.com/raspberry-pi-high-quality-camera-autofocus-motorized-focus-imx477.html). 

AF expects the lens control to come through the kernel as a V4L2 subdevice though.
I do have a project on the go that adds an RP2040 as either stepper motor control for lenses such as https://www.ebay.co.uk/itm/115865242506, or motor and potentiometer control for CCTV lenses (I've got an old [Computar 7.5-120mm lens](https://computar.fr/en/products/lenses/zoom-lenses/motor-zoom/h16z7516pdc-130.html)). The interface to the RP2040 would be via a V4L2 sub-device driver. At some point I might get back to it.

A question for @naushir or @davidplowman would be whether FoM metadata should always be added, irrespective of whether libcamera has a lens device to control or not. (I'll confess to not having followed the code to ensure that it isn't added already).

davidplowman, 2025-02-20:
Down in libcamera we certainly always add the focus FoM, whether or not there is a lens that we know about. It's pretty essential for manually focusing the lens on the HQ cam, for example, so something like gstreamer (not that I know anything about it) should probably be making that available to applications too.

6by9, 2025-02-20:
> I don't want to appear rude or uninterested and thank you for participating. I first have to understand what that means now and how I can move forward. The fact is, the Arduinos have a imx477af.json implementation.

I assume that's a typo for Arducam. https://github.com/ArduCAM/libcamera/blob/arducam/src/ipa/rpi/vc4/data/imx477_af.json

In which case we direct you to Arducam for support. They've chosen not to work with either mainline libcamera or Raspberry Pi in upstreaming their configuration files, therefore the support burden is on them.

https://forums.raspberrypi.com/viewtopic.php?t=372520