raspberrypi/libcamera #243: Kernel 6.12 Compatibility

schoolpost, 2025-03-03:
Certainly jumping the gun on this one, I'm attempting to get rpicam-apps/libcamera working with the IMX294 on kernel `6.12`

I've had to make some minor adjustments to the driver to account for some v4l2 api changes, specifically:
`v4l2_subdev_get_try_crop` -> `v4l2_subdev_state_get_crop`
`v4l2_subdev_get_try_format` ->` v4l2_subdev_state_get_format`

and:
`#include <asm/unaligned.h>` -> `#include <linux/unaligned.h>`

With that I can get the driver to successfully compile. However there are new issues to contend with. I have image artifacts for nearly all the sensor modes. 
```    SRGGB10_CSI2P,2088x1096/241.255 - Score: 12509.4
    SRGGB10_CSI2P,1896x1404/190.223 - Score: 14202.4
    SRGGB12_CSI2P,1392x248/461.681 - Score: 0
    SRGGB12_CSI2P,1392x736/351.494 - Score: 11286.8
    SRGGB12_CSI2P,3872x2180/61.4288 - Score: 12613.3
    SRGGB12_CSI2P,4144x2184/57.7601 - Score: 12318.4
    SRGGB12_CSI2P,4176x2184/54.3685 - Score: 12282.4
    SRGGB12_CSI2P,3792x2840/41.8305 - Score: 14081.1
    SRGGB14_CSI2P,3792x2840/28.8218 - Score: 15081.1
```

I say nearly because there are 2 odd sensor modes that seem to not be affected...for whatever reason. 

12-bit 1392x248:
[image]

14-bit 3792x2840:
[image]

All the other modes exhibit some kind of garbled image output:

12-bit 1392x736
[image]

12-bit 4144x2184
[image]

Certainly something not right...but I'm not sure where specifically that is ( PiSP, CFE, Libcamera, elsewhere? )

schoolpost, 2025-03-03:
> Is the corruption present in the RAW/Bayer data for the 12-bit case? If so I would suggest looking at the sensor driver register set as a start.

Yes, the RAW is affected. I'm very confused why there are 2 modes that work without issue, and more so confused that the driver is otherwise artifact free and functional in kernel `6.6` 

Is there anywhere specific I should look first within the driver?

naushir, 2025-03-03:
One obvious thing to check is the CSI-2 link frequency is calculated correctly.  The code for that can be found here:

here is a

6by9, 2025-03-03:
The 12-bit 1392x736 corruption looks like some FIFO overflows, or the sensor stops early and you're getting uninitialised memory.

Link frequency is the most likely candidate. What does `v4l2-ctl --list-ctrls-menu -d /dev/v4l-subdev2` list as the controls? (correct the subdev if necessary)

schoolpost, 2025-03-04:
This appears to be in an issue of RP1 clock speed. To achieve 4K60 capability we have to run a mild overclock of RP1 ( bump it 300Mhz for ~580Mpx/s and modifying:  `.minPixelProcessingTime = 1.0us / 580,` ) 

Doesn't look like `6.12` is willing to accept the higher assigned clock, is there new procedure for achieving this?

naushir, 2025-03-04:
It's been a while since I looked at overclocking RP1.  What runes are you using to set the clock speed to 300MHz?

schoolpost, 2025-03-04:
Found here: https://github.com/raspberrypi/linux/blob/rpi-6.6.y/arch/arm64/boot/dts/broadcom/rp1.dtsi

Changing both `<200000000>,  // RP1_PLL_SYS` & `<200000000>,  // RP1_CLK_SYS`

naushir, 2025-03-04:
Do you have a log from the clock driver when setting up clk_sys for the 6.12 tree by any chance?

schoolpost, 2025-03-04:
I don't currently, but as I recall scanning through `dmesg` earlier, I didn't see any explicit logging related to clk setting unlike what I saw in `6.6` : `[    0.530488] clk: couldn't set clk_sys clk rate to 300000000 (-22), current rate: 333333333`

naushir, 2025-03-04:
I'll take a look and see what I can find.  Unfortunately I'm a bit swamped with other things, so it may a while before I get to this.

schoolpost, 2025-03-04:
I've jumped back to `6.6` in the mean time, but I'd love to get onto `6.12` as soon as I can.

fiepfiep, 2025-03-11:
for me 6.12 works just fine..

closed 2025-03-22 by schoolpost