raspberrypi/libcamera #244: Arducam 0v5647, RaspberryPi5 + Ubuntu24: No cameras detected

bryangd34, 2025-03-05:
I installed libcamera and have my /boot/firmware/config.txt updated with:
camera_auto_detect=0
dtoverlay=ov5647

I am also experimented with:
dtoverlay=ov5647,cam0

`LIBCAMERA_LOG_LEVELS=*:DEBUG cam -l`

> [19:37:25.103720641] [22622] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_pisp.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so is signed
> [19:37:25.103831513] [22622] DEBUG IPAManager ipa_manager.cpp:235 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so'
> [19:37:25.103943255] [22622] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_vc4.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so is signed
> [19:37:25.103972792] [22622] DEBUG IPAManager ipa_manager.cpp:235 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so'
> [19:37:25.103983218] [22622]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+53-29156679
> [19:37:25.104147571] [22625] DEBUG Camera camera_manager.cpp:73 Starting camera manager
> [19:37:25.112905449] [22625] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media3
> [19:37:25.112923376] [22625] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: rp1-cfe
> [19:37:25.113072933] [22625] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpivid" created from /dev/media2
> [19:37:25.113086803] [22625] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 1 missing dependencies
> [19:37:25.113332157] [22625] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
> [19:37:25.113342287] [22625] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: rpivid
> [19:37:25.113481122] [22625] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media0
> [19:37:25.113492751] [22625] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 9 missing dependencies
> [19:37:25.113625845] [22625] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media1
> [19:37:25.113636919] [22625] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 9 missing dependencies
> [19:37:25.115518085] [22625] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
> [19:37:25.115529567] [22625] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: pispbe
> [19:37:25.117395491] [22625] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
> [19:37:25.117406029] [22625] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: pispbe
> [19:37:25.117679031] [22625] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'rpi/vc4'
> [19:37:25.117717476] [22625] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
> [19:37:25.117727772] [22625] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
> [19:37:25.117741198] [22625] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'rpi/pisp'
> [19:37:25.117755254] [22625] DEBUG RPI pisp.cpp:890 Unable to acquire a CFE instance
> Available cameras:

I suspect it is a hardware issue, e.g. bad connection. But cannot check that at the moment as I am not on-site.

Are there any other (software) debugging checks you could recommend to help narrow down the issue please?
I will be asking someone soon who is on-site to double check the connecting cable.

Thanks in advance.

BGD

naushir, 2025-03-05:
Could you post the output of ``rpicam-hello --version`` please?

6by9, 2025-03-05:
Check your kernel logs for error messages from probing the sensor. (`sudo dmesg`)

bryangd34, 2025-03-05:
@6by9 and @naushir thank you for your suggestions!
The rig has been powered down. I'll look as soon as it's back up and report back.

bryangd34, 2025-03-06:
Cable was the wrong way around!