raspberrypi/libcamera #246: 'AeLocked' is not a member of libcamera::controls

kartheekmedathati, 2025-03-11:
Encountering the above error :
[Processing: camera_ros]                                 
[ 46%] Building CXX object CMakeFiles/param.dir/src/types.cpp.o
[ 53%] Building CXX object CMakeFiles/param.dir/src/type_extent.cpp.o
/ros2_modules/camera_ros/src/camera_ros/src/type_extent.cpp: In function âstd::size_t get_extent(const libcamera::ControlId*)â:
/home/medathati/ros2_modules/camera_ros/src/camera_ros/src/type_extent.cpp:42:6: error: âAeLockedâ is not a member of âlibcamera::controlsâ; did you mean âAwbLockedâ?
   42 |   IF(AeLocked)
      |      ^~~~~~~~
/ros2_modules/camera_ros/src/camera_ros/src/type_extent.cpp:33:40: note: in definition of macro âIFâ
   33 |   if (id->id() == libcamera::controls::T.id()) \

FrederikLauber, 2025-03-12:
Same here I guess:

`INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /usr/bin/ninja -C /home/alarm/rpicam-apps/build -j 1
ninja: Entering directory `/home/alarm/rpicam-apps/build'
[5/42] Compiling C++ object rpicam_app.so.1.6.0.p/core_rpicam_app.cpp.o
FAILED: rpicam_app.so.1.6.0.p/core_rpicam_app.cpp.o
c++ -Irpicam_app.so.1.6.0.p -I. -I.. -I/usr/local/include/libcamera -I/usr/include -I/usr/include/libpng16 -I/usr/include/libdrm -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -Wall -Winvalid-pch -Wextra -Wpedantic -Werror -std=c++17 -O3 -pedantic -Wno-unused-parameter -faligned-new -D_FILE_OFFSET_BITS=64 -Wno-psabi -DLIBDRM_PRESENT=1 -fPIC -pthread -DBOOST_ALL_NO_LIB -MD -MQ rpicam_app.so.1.6.0.p/core_rpicam_app.cpp.o -MF rpicam_app.so.1.6.0.p/core_rpicam_app.cpp.o.d -o rpicam_app.so.1.6.0.p/core_rpicam_app.cpp.o -c ../core/rpicam_app.cpp
In file included from ../core/rpicam_app.cpp:10:
../core/frame_info.hpp: In constructor ▒FrameInfo::FrameInfo(const CompletedRequestPtr&)▒:
../core/frame_info.hpp:50:58: error: ▒AeLocked▒ is not a member of ▒libcamera::controls▒; did you mean ▒AwbLocked▒?
   50 |                 auto ae = ctrls.get(libcamera::controls::AeLocked);
      |                                                          ^~~~~~~~
      |                                                          AwbLocked
ninja: build stopped: subcommand failed.
`

I am trying to compile on a Raspberry pi 3 running Arm Arch Linux armv7l.
libcamera is the current master (39419ce431dbd4f34d8772bd31bb7f44a3534f86)

naushir, 2025-03-17:
The latest commit of libcamera is at https://github.com/raspberrypi/libcamera/commit/29156679717bec7cc4784aeba3548807f2c27fca (release [v0.4.0+rpt20250213](https://github.com/raspberrypi/libcamera/releases/tag/v0.4.0%2Brpt20250213))   Can you please update your local libcamera installation to this release and you should not get this problem.

FrederikLauber, 2025-03-17:
Thanks for your response. I think I figured out what was going on on my site. I was following https://blog.jirkabalhar.cz/2024/02/raspberry-camera-on-archlinux-arm-in-2024/

What I did not realize was, that the libcamera is taken from upstream while the rpicam-apps are taken from here. The commit I quotes is from be begin of March so I guess stuff was merged upstream which results in a build error. I will try again with libcamera from here and report back.

FrederikLauber, 2025-03-17:
Using libcamera from this repo and not from upstream works. Not sure if my issue is the same as @kartheekmedathati so I will not close it.

closed 2025-04-01 by naushir