raspberrypi/libcamera #250: Build fails with gcc15

ggardet, 2025-03-20:
Build fails with gcc15.
Error log:
```
[   25s] FAILED: src/libcamera/libcamera.so.0.4.0.p/dma_buf_allocator.cpp.o 
[   25s] c++ -Isrc/libcamera/libcamera.so.0.4.0.p -Isrc/libcamera -I../src/libcamera -Iinclude -I../include -Iinclude/libcamera -Iinclude/libcamera/ipa -Iinclude/libcamera/internal -Isrc/libcamera/proxy -I/usr/include/p11-kit-1 -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Werror -std=c++17 -Wno-redundant-move -Wmissing-declarations -Wshadow -include /home/abuild/rpmbuild/BUILD/libcamera-0.4.0-build/libcamera-0.4.0/x86_64-suse-linux/config.h -O2 -Wall -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=3 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -g -fPIC -DLIBCAMERA_BASE_PRIVATE -MD -MQ src/libcamera/libcamera.so.0.4.0.p/dma_buf_allocator.cpp.o -MF src/libcamera/libcamera.so.0.4.0.p/dma_buf_allocator.cpp.o.d -o src/libcamera/libcamera.so.0.4.0.p/dma_buf_allocator.cpp.o -c ../src/libcamera/dma_buf_allocator.cpp
[   25s] In file included from ../src/libcamera/dma_buf_allocator.cpp:9:
[   25s] ../include/libcamera/internal/dma_buf_allocator.h:66:19: error: ‘uint64_t’ has not been declared
[   25s]    66 |         void sync(uint64_t step);
[   25s]       |                   ^~~~~~~~
[   25s] ../include/libcamera/internal/dma_buf_allocator.h:17:1: note: ‘uint64_t’ is defined in header ‘<cstdint>’; this is probably fixable by adding ‘#include <cstdint>’
[   25s]    16 | #include <libcamera/base/unique_fd.h>
[   25s]   +++ |+#include <cstdint>
[   25s]    17 | 
[   25s] ../include/libcamera/internal/dma_buf_allocator.h:69:9: error: ‘uint64_t’ does not name a type
[   25s]    69 |         uint64_t flags_ = 0;
[   25s]       |         ^~~~~~~~
[   25s] ../include/libcamera/internal/dma_buf_allocator.h:69:9: note: ‘uint64_t’ is defined in header ‘<cstdint>’; this is probably fixable by adding ‘#include <cstdint>’
[   25s] ../src/libcamera/dma_buf_allocator.cpp: In constructor ‘libcamera::DmaSyncer::DmaSyncer(libcamera::SharedFD, SyncType)’:
[   25s] ../src/libcamera/dma_buf_allocator.cpp:301:17: error: ‘flags_’ was not declared in this scope; did you mean ‘Flags’?
[   25s]   301 |                 flags_ = DMA_BUF_SYNC_READ;
[   25s]       |                 ^~~~~~
[   25s]       |                 Flags
[   25s] ../src/libcamera/dma_buf_allocator.cpp: At global scope:
[   25s] ../src/libcamera/dma_buf_allocator.cpp:319:6: error: no declaration matches ‘void libcamera::DmaSyncer::sync(uint64_t)’
[   25s]   319 | void DmaSyncer::sync(uint64_t step)
[   25s]       |      ^~~~~~~~~
[   25s] ../include/libcamera/internal/dma_buf_allocator.h:66:14: note: candidate is: ‘void libcamera::DmaSyncer::sync(int)’
[   25s]    66 |         void sync(uint64_t step);
[   25s]       |              ^~~~
[   25s] ../include/libcamera/internal/dma_buf_allocator.h:52:7: note: ‘class libcamera::DmaSyncer’ defined here
[   25s]    52 | class DmaSyncer final
[   25s]       |       ^~~~~~~~~
```

ggardet, 2025-03-20:
Adding `#include <cstdint>` should be enough.

naushir, 2025-03-20:
Since this is part of the core libcamera source files, may I suggest that you post this to the [libcamera-devel](https://lists.libcamera.org/listinfo/libcamera-devel) mailing list please?  Once this is fixed upstream, I will merge the patch to our downstream tree.

closed 2025-03-20 by naushir