raspberrypi/libcamera #255: Camera Stream error

sowmyasammy-developer, 2025-04-11:
I am trying to capture image and streaming using mira220 sensor. i connected my camera to raspberrypi 4 and powered it by connecting to my laptop. After adding required configuration files and drivers camera is getting detected. But when i try to capture an image i am getting an error.
I am a beginner just started working with camera modules. I am attaching the error logs. Can someone help me what could be the issue.

LOGS
---------------------------------------------------------------------------------------------------------------------------------------------
/libcamera-apps/build $ /usr/local/bin/libcamera-hello --list-cameras
Available cameras
-----------------
0 : mira220 [1600x1400 12-bit GRBG] (/base/soc/i2c0mux/i2c@1/mira220@54)
    Modes: 'SGRBG10_CSI2P' : 640x480 [172.06 fps - (480, 460)/640x480 crop]
                             1600x1400 [60.27 fps - (0, 0)/1600x1400 crop]
           'SGRBG12_CSI2P' : 640x480 [172.06 fps - (480, 460)/640x480 crop]
                             1600x1400 [60.27 fps - (0, 0)/1600x1400 crop]
           'SGRBG8' : 640x480 [172.06 fps - (480, 460)/640x480 crop]
                      1600x1400 [60.27 fps - (0, 0)/1600x1400 crop]

--------------------------
/usr/local/bin/libcamera-jpeg -o test.jpg
[0:22:48.849839377] [5150]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+56-b7af1b4e
[0:22:48.863711479] [5151]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:22:48.865463230] [5151]  WARN RPI vc4.cpp:393 Mismatch between Unicam and CamHelper for embedded data usage!
[0:22:48.866313764] [5151]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/mira220@54 to Unicam device /dev/media1 and ISP device /dev/media4
[0:22:48.866373504] [5151]  INFO RPI pipeline_base.cpp:1121 Using configuration file '/usr/local/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Preview window unavailable
Mode selection for 800:700:12:P
    SGRBG10_CSI2P,640x480/0 - Score: 1831.43
    SGRBG10_CSI2P,1600x1400/0 - Score: 1375
    SGRBG12_CSI2P,640x480/0 - Score: 831.429
    SGRBG12_CSI2P,1600x1400/0 - Score: 375
    SGRBG8,640x480/0 - Score: 2831.43
    SGRBG8,1600x1400/0 - Score: 2375
Stream configuration adjusted
[0:22:48.868325013] [5150]  INFO Camera camera.cpp:1202 configuring streams: (0) 800x700-YUV420 (1) 1600x1400-SGRBG12_CSI2P
[0:22:48.868816678] [5151]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/mira220@54 - Selected sensor format: 1600x1400-SGRBG12_1X12 - Selected unicam format: 1600x1400-pgCC
[0:22:49.107859771] [5151] ERROR V4L2 v4l2_videodevice.cpp:2049 /dev/video0[12:cap]: Failed to start streaming: Remote I/O error
ERROR: *** failed to start camera ***

---------------------------------------------------------------------------------------------------------------------------------------------

[image]
[image]

6by9, 2025-04-11:
`Remote I/O error` aka EREMOTEIO is generally the result of an I2C transfer failure. Check your driver and the kernel logs.

6by9, 2025-04-15:
Your driver has told you the bit that failed
```
[  697.386716] mira220 10-0054: Could not write stream-on sequence
```
The mira220 driver is not in our kernel, therefore not our responsibility. Please ask those who have supplied the driver to you.

There is a draft PR at https://github.com/raspberrypi/linux/pull/6717. Whilst it does have a `mira220_identify_module` function, it never checks for errors, so will allow the sensor to probe fully even if not correctly configured or connected.

naushir, 2025-04-25:
Closing this now as it's not related to libcamera, rather it seems to be a problem with HW or the kernel drive.

closed 2025-04-25 by naushir