raspberrypi/libcamera #256: [Pi5] Pixel format not respected at validation & during opening

Moddingear, 2025-04-12:
I'm on the Pi5, using libcamera from C++ with two ov9281 (monochrome global shutter camera) (from arducam).
Those cameras have two pixel formats : R8 and R10_CSI2P (as listed within libcamera and using `libcamera-hello --list-cameras`.

So when I want to open the camera, and I ask for R8, I get R16, and when I ask for R10_CSI2P, I get MONO_PISP_COMP1, which makes absolutely no sense.

Here's how I'm opening the camera :
```cpp
	cout << "Requested for " << GetName() << " configuration is " << setting.toString() << endl;
	config = camera->generateConfiguration( { StreamRole::Raw } );
	StreamConfiguration &streamConfig = config->at(0);
	const auto &formats = streamConfig.formats();
	int found = 0;
	libcamera::PixelFormat pf;
	for (auto pix : formats.pixelformats())
	{
		if (pix.fourcc() == setting.fourcc)
		{
			cout << "Found pixel format: " << pix << endl;
			pf = pix;
			found |= 1;
			break;
		}
	}
	for (auto size : formats.sizes(pf))
	{
		SensorMode mode(size, pf, 0);
		if (size.height == setting.height && size.width == setting.width)
		{
			cout << "Found size/mode: " << mode.ToString() << "d" << mode.depth() << endl;
			config->at(0).size = size;
			config->at(0).pixelFormat = pf;
			config->sensorConfig = SensorConfiguration();
			config->sensorConfig->outputSize = size;
			config->sensorConfig->bitDepth = mode.depth();
			found |= 10;
			break;
		}
	}
	if (found != 11)
	{
		cerr << "Failed to find some config: " << found << endl;
	}
	
	config->validate();
	cout << "Validated configuration is " << streamConfig.toString() << " (" << streamConfig.pixelFormat << ")" << endl;
	camera->configure(config.get());
	cout << "Post configuration is " << streamConfig.toString() << " (" << streamConfig.pixelFormat << ")" << endl;
	allocator = make_unique<FrameBufferAllocator>(camera);
	for (StreamConfiguration &cfg : *config) {
		int ret = allocator->allocate(cfg.stream());
		if (ret < 0) {
			std::cerr << "Can't allocate buffers" << std::endl;
			return;
		}
		size_t allocated = allocator->buffers(cfg.stream()).size();
		std::cout << "Allocated " << allocated << " buffers for stream" << std::endl;
	}
```
which prints this :
```
Requested for /base/axi/pcie@120000/rp1/i2c@88000/ov9281@60 configuration is MONO R10  1280x800@120.76
Found pixel format: R10_CSI2P
Found size/mode: R10_CSI2P,1280x800/0d10
Validated configuration is 1280x800-MONO_PISP_COMP1 (MONO_PISP_COMP1)
Post configuration is 1280x800-MONO_PISP_COMP1 (MONO_PISP_COMP1)
Allocated 2 buffers for stream
```
or
```
Requested for /base/axi/pcie@120000/rp1/i2c@80000/ov9281@60 configuration is MONO R8   1280x800@150.97
Found pixel format: R8
Found size/mode: R8,1280x800/0d8
Validated configuration is 1280x800-R16 (R16)
Post configuration is 1280x800-R16 (R16)
Allocated 2 buffers for stream
```
And so, when I want 8 bits per pixel, I get 16, and when I want 10 I get 8 💢 .

`libcamera-hello --list-cameras` prints this:
```
Available cameras
-----------------
0 : ov9281 [1280x800 10-bit MONO] (/base/axi/pcie@120000/rp1/i2c@88000/ov9281@60)
    Modes: 'R8' : 640x400 [325.52 fps - (0, 0)/1280x800 crop]
                  1280x720 [180.51 fps - (0, 0)/1280x720 crop]
                  1280x800 [150.97 fps - (0, 0)/1280x800 crop]
           'R10_CSI2P' : 640x400 [260.42 fps - (0, 0)/1280x800 crop]
                         1280x720 [144.40 fps - (0, 0)/1280x720 crop]
                         1280x800 [120.76 fps - (0, 0)/1280x800 crop]

1 : ov9281 [1280x800 10-bit MONO] (/base/axi/pcie@120000/rp1/i2c@80000/ov9281@60)
    Modes: 'R8' : 640x400 [325.52 fps - (0, 0)/1280x800 crop]
                  1280x720 [180.51 fps - (0, 0)/1280x720 crop]
                  1280x800 [150.97 fps - (0, 0)/1280x800 crop]
           'R10_CSI2P' : 640x400 [260.42 fps - (0, 0)/1280x800 crop]
                         1280x720 [144.40 fps - (0, 0)/1280x720 crop]
                         1280x800 [120.76 fps - (0, 0)/1280x800 crop]
```
This is using the current main branch.

6by9, 2025-04-12:
You're conflating sensor output format vs CSI2 receiver output format.

The CFE (Pi5 CSI2 receiver) will always output 16bpc (left shifting the data) or the compressed formats.

`rpicam-hello --viewfinder-mode 1280:800:10:U` vs `rpicam-hello --viewfinder-mode 1280:800:10:P`should provide details of the relevant config options required to request the two.

Moddingear, 2025-04-12:
Running the commands you listed, I get this :

```
>>> rpicam-hello --viewfinder-mode 1280:800:10:P
[0:01:55.607382830] [1173]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+53-29156679
[0:01:55.619611731] [1174]  INFO RPI pisp.cpp:720 libpisp version v1.0.7 28196ed6edcf 18-03-2025 (19:00:09)
[0:01:55.624212710] [1174]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/ov9281@60 to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[0:01:55.624301325] [1174]  INFO RPI pisp.cpp:720 libpisp version v1.0.7 28196ed6edcf 18-03-2025 (19:00:09)
[0:01:55.627574000] [1174]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/ov9281@60 to CFE device /dev/media4 and ISP device /dev/media1 using PiSP variant BCM2712_C0
Preview window unavailable
Mode selection for 1280:800:10:P
    R8,640x400/0 - Score: 3080
    R8,1280x720/0 - Score: 1226.67
    R8,1280x800/0 - Score: 1000
    R10_CSI2P,640x400/0 - Score: 2080
    R10_CSI2P,1280x720/0 - Score: 226.667
    R10_CSI2P,1280x800/0 - Score: 0
Stream configuration adjusted
[0:01:55.631880466] [1173]  INFO Camera camera.cpp:1202 configuring streams: (0) 640x400-YUV420 (1) 1280x800-MONO_PISP_COMP1
[0:01:55.631996397] [1174]  INFO RPI pisp.cpp:1484 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/ov9281@60 - Selected sensor format: 1280x800-Y10_1X10 - Selected CFE format: 1280x800-PC1M
[0:01:55.636904317] [1178]  WARN IPARPI ipa_base.cpp:1114 Could not set SHARPNESS - no sharpen algorithm```
```
>>> rpicam-hello --viewfinder-mode 1280:800:10:U
[0:01:00.434057001] [857]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+53-29156679
[0:01:00.451237446] [874]  INFO RPI pisp.cpp:720 libpisp version v1.0.7 28196ed6edcf 18-03-2025 (19:00:09)
[0:01:00.521320284] [874]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/ov9281@60 to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[0:01:00.521417804] [874]  INFO RPI pisp.cpp:720 libpisp version v1.0.7 28196ed6edcf 18-03-2025 (19:00:09)
[0:01:00.525028744] [874]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/ov9281@60 to CFE device /dev/media4 and ISP device /dev/media1 using PiSP variant BCM2712_C0
Preview window unavailable
Mode selection for 1280:800:10:U
    R8,640x400/0 - Score: 3080
    R8,1280x720/0 - Score: 1226.67
    R8,1280x800/0 - Score: 1000
    R10_CSI2P,640x400/0 - Score: 2080
    R10_CSI2P,1280x720/0 - Score: 226.667
    R10_CSI2P,1280x800/0 - Score: 0
Stream configuration adjusted
[0:01:00.529589678] [857]  INFO Camera camera.cpp:1202 configuring streams: (0) 640x400-YUV420 (1) 1280x800-R16
[0:01:00.529715236] [874]  INFO RPI pisp.cpp:1484 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/ov9281@60 - Selected sensor format: 1280x800-Y10_1X10 - Selected CFE format: 1280x800-Y16 
[0:01:00.532735408] [880]  WARN IPARPI ipa_base.cpp:1114 Could not set SHARPNESS - no sharpen algorithm```

I'm only interested in raw sensor data, not viewfinder (since I don't have any bayer grid). Running those commands doesn't give me any insight, as both commands get the same pixel format...

6by9, 2025-04-12:
```
Selected sensor format: 1280x800-Y10_1X10 - Selected CFE format: 1280x800-PC1M
```
vs 
```
Selected sensor format: 1280x800-Y10_1X10 - Selected CFE format: 1280x800-Y16 
```
which is as expected.

Ask for 1280:800:8:P and you should get 
```
Selected sensor format: 1280x800-Y8_1X8 - Selected CFE format: 1280x800-PC1M
```
and 1280:800:8:U
```
Selected sensor format: 1280x800-Y8_1X8 - Selected CFE format: 1280x800-Y16
```

Moddingear, 2025-04-13:
It's not quite what you predicted:
```
>>> rpicam-hello --viewfinder-mode 1280:800:8:P
configuring streams: (0) 640x400-YUV420 (1) 1280x800-R16
Selected sensor format: 1280x800-Y8_1X8 - Selected CFE format: 1280x800-Y16 
```
```
rpicam-hello --viewfinder-mode 1280:800:8:U
configuring streams: (0) 640x400-YUV420 (1) 1280x800-R16
Selected sensor format: 1280x800-Y8_1X8 - Selected CFE format: 1280x800-Y16 
```

But that's besides my point. Why can't I get the raw data in 8 bits per pixel when I want it? There's no documentation (that I could find) about this behaviour. I don't want YUV as I don't have any colors, and I don't want 8 bits expanded to 16bits, or 10bits truncated to 8bits.

Also PC1M format doesn't even exist in the library (formats.h)...

6by9, 2025-04-14:
As I have already stated, the CFE (Pi5 CSI2 receiver) will always output 16bpc (left shifting the data) or the compressed formats.

There are four alternate output paths that do write data directly from the CSI-2 interface. One is used for the embedded metadata where appropriate.
libcamera does not use these paths for writing image data as it wants the statistics that are also produced by the CFE for AE/AGC/AWB etc.

You can choose to not use libcamera and get the raw data streams. This is what is done for the TC358743 HDMI to CSI2 bridge and ADV7282-MA analogue video to CSI2 bridges. You will need to configure the pipelines appropriately via the Media Controller API. There are several forum posts covering that for those bridge chips.

Documentation of the CFE is in the PiSP specification  - https://datasheets.raspberrypi.com/camera/raspberry-pi-image-signal-processor-specification.pdf. Sections 3 and 6 are probably the main ones you're after.

naushir, 2025-04-23:
As mentioned by @6by9, the CFE will remap 8 -> 16 bits unconditionally in hardware, and this path will be taken if you use libcamera.  If you want 8-bits unconditionally, you can run the sensor directly through V4L2 and the media controller.

Closing now, but please do reopen if you have further questions.

closed 2025-04-23 by naushir