raspberrypi/libcamera #258: IPC and IPA errors on RPi 5 Bookworm

fbridge8, 2025-04-25:
Hello everybody.

I'm having issues with libcamera, somewhat similar to these other issues:

https://github.com/raspberrypi/libcamera/issues/254 - Even the sensor is the same VD56G3, but I'm not using calibration files.
https://github.com/raspberrypi/libcamera/issues/115 - This didn't have a resolution, but I have installed `libgnutls28-dev` (I've followed the _exact_ steps in the README to build, but same issue happen with libcamera's system package).

I went through some extensive debug in this other issue on `camera_ros` https://github.com/christianrauch/camera_ros/issues/115

Basically this is what's happening (more detailed logs below).

```bash
[10:02:34.005492925] [181325] ERROR IPCPipe ipc_pipe_unixsocket.cpp:134 Call timeout!
[10:02:34.005523463] [181325] ERROR IPCPipe ipc_pipe_unixsocket.cpp:66 Failed to call sync
[10:02:34.005531333] [181325] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:261 Failed to call start
[10:02:34.009190798] [181325] ERROR IPCUnixSocket ipc_unixsocket.cpp:192 Failed to send: Connection refused
[10:02:34.009221799] [181325] ERROR IPCPipe ipc_pipe_unixsocket.cpp:66 Failed to call sync
[10:02:34.009228836] [181325] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:432 Failed to call mapBuffers
[10:02:34.067543539] [181325] ERROR IPCUnixSocket ipc_unixsocket.cpp:192 Failed to send: Transport endpoint is not connected
[10:02:34.067573243] [181325] ERROR IPCPipe ipc_pipe_unixsocket.cpp:80 Failed to call async
[10:02:34.067580317] [181325] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:508 Failed to call prepareIsp
```

```bash
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.12.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
```

<details><summary>(expand) LIBCAMERA_LOG_LEVELS=*:DEBUG when running the ROS2 node</summary>
<p>

```bash
pi@raspberrypi:~ $ LIBCAMERA_LOG_LEVELS=*:DEBUG ros2 run --prefix 'gdb -ex run --args' camera_ros camera_node --ros-args --log-level camera:=debug -p camera:=1 -p width:=640 -p height:=480GNU gdb (Debian 13.1-3) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/pi/ros2_jazzy/install/camera_ros/lib/camera_ros/camera_node...
(No debugging symbols found in /home/pi/ros2_jazzy/install/camera_ros/lib/camera_ros/camera_node)
Starting program: /home/pi/ros2_jazzy/install/camera_ros/lib/camera_ros/camera_node --ros-args --log-level camera:=debug -p camera:=1 -p width:=640 -p height:=480
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff614a040 (LWP 181242)]
[New Thread 0x7ffff593a040 (LWP 181243)]
[New Thread 0x7fffefb7e040 (LWP 181244)]
[New Thread 0x7fffef36e040 (LWP 181245)]
[New Thread 0x7fffde4ae040 (LWP 181246)]
[New Thread 0x7fffdbc9e040 (LWP 181247)]
[New Thread 0x7fffd948e040 (LWP 181248)]
[New Thread 0x7fffd6c7e040 (LWP 181249)]
[New Thread 0x7fffd63e6040 (LWP 181250)]
[New Thread 0x7fffd5bd6040 (LWP 181251)]
[New Thread 0x7fffd53c6040 (LWP 181252)]
[... 593 lines cut ...]
[9:49:29.859410376] [181259] DEBUG V4L2 v4l2_videodevice.cpp:1731 /dev/video7[58:out]: Dequeuing buffer 3
[9:49:29.859430450] [181259] DEBUG RPI pisp.cpp:1712 Stream CFE Config buffer dequeue, buffer id 4, timestamp: 35369850072000
[9:49:29.859446802] [181259] DEBUG RPI pipeline_base.cpp:1442 Returning buffer to stream CFE Config
[9:49:29.859463265] [181259] DEBUG RPISTREAM rpi_stream.cpp:271 Queuing buffer 6 for CFE Config
[9:49:29.859482339] [181259] DEBUG V4L2 v4l2_videodevice.cpp:2011 /dev/video7[58:out]: Queueing buffer 5
[9:49:29.859554006] [181259] DEBUG V4L2 v4l2_videodevice.cpp:1731 /dev/video6[59:cap]: Dequeuing buffer 3
[9:49:29.859575858] [181259] DEBUG RPI pisp.cpp:1712 Stream CFE Stats buffer dequeue, buffer id 4, timestamp: 35369850072000
```

</p>
</details> 

There are some signature issues above, not sure if related.

```bash
DEBUG IPAManager ipa_manager.cpp:306 IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so signature is not valid
```

Because they also appear in `LIBCAMERA_LOG_LEVELS=*:DEBUG libcamera-hello -t 100` but it doesn't fail there.

[libcamera_debug_logs.txt](https://github.com/user-attachments/files/19905307/libcamera_debug_logs.txt)

Any help/direction is welcome.

Thanks!

kbingham, 2025-04-25:
As with https://github.com/raspberrypi/libcamera/issues/254#issuecomment-2809706209 - make sure you install all the dependencies - even though they are marked optional for IPA signing.

If the IPA is run externally - and crashes - currently you don't get any notification of what the crash is (which is really unhelpful).

There's some patches to get the errors for external IPAs reported in the libcamera logs - but they aren't merged/upstream yet.

So the easiest way to be able to see what went wrong is to be absolutely sure that the IPA is not isolated - which means install the dependencies - and make sure the line "DEBUG IPAManager ipa_manager.cpp:306 IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so signature is not valid" is not printed. That line means the IPA is going to be run in isolation (and you won't get the IPA logs).

kbingham, 2025-04-25:
Except- I think I just read in your 'other' issues that you have installed an /external/ IPA package from ST.

"Only libcamera-ipa:arm64 0.4.0+rpt20250213+st20250401-1 comes from ST."

You will really have to check what ST are doing there if you're not compiling this yourself from source.

naushir, 2025-04-25:
Indeed, you definately don't want to run in IPA isolated mode on the RPi platforms as there is no reason to.  Do you use the Raspberry Pi provided libcamera/ipa library files?  If so, you really should not be running into this as they will be signed correctly at build time before packaging.

kbingham, 2025-04-25:
I've pinged ST on this directly in the libcamera IRC channel ...

fbridge8, 2025-04-25:
Yes, getting the IPA package from https://www.st.com/en/embedded-software/stsw-img502.html

What confuses me is that running `libcamera-hello` also raises the same "signature is not valid" I get when running ROS, but doesn't crash, and produces images as if nothing had happened. How's that possible?

I'll raise this in the ST forum. Any direction or something I should mention to better steer them towards the issue?

naushir, 2025-04-25:
> Any direction or something I should mention to better steer them towards the issue?

I suggest mentioning that the IPA isolation (due to invalid signature) hides any error messages (for now) making debug difficult, and the best outcome would be to upstream code to support this sensor so we don't run in isolated mode.

kbingham, 2025-04-25:
"Call timeout!" is the same as saying "The IPA has crashed or failed to load".

Figuring that out - requires getting the logs from the IPA... hopefully - someone at ST can do this.

mugnierb, 2025-04-25:
If you open an issue on ST forums please put a link here so we can follow along at ST ;)

estevanmaito, 2025-04-25:
We've opened an issue on ST forum, here's a link for reference: https://community.st.com/t5/imaging-sensors/invalid-signature-error-in-ipa-module-for-vd56g3/m-p/796819

Many thanks to everybody involved.