raspberrypi/libcamera #263: Failing to Find jinja2

meawoppl, 2025-05-17:
I'm seeing the build process fail to find jinja2 despite it being installed and reported as detected in the meson outputs:

```
cd libcamera &&     git submodule update --init &&     meson setup build --buildtype=release -Dpipelines=rpi/vc4,rpi/pisp -Dipas=rpi/vc4,rpi/pisp -Dv4l2=true -Dgstreamer=enabled -Dtest=false -Dlc-compliance=disabled -Dcam=disabled -Dqcam=disabled -Ddocumentation=disabled -Dpycamera=enabled &&     cd build &&     ninja &&     ninja install
[camera]    ---> Running in 5757fdd6a9fc
[camera]   Cloning into 'libpisp'...
[camera]   
[camera]   The Meson build system
[camera]   Version: 1.3.2
[camera]   Source dir: /libcamera
[camera]   Build dir: /libcamera/build
[camera]   Build type: native build
[camera]   DEPRECATION: Option 'v4l2' value 'true' is replaced by 'enabled'
[camera]   Project name: libcamera
[camera]   Project version: 0.5.0
[camera]   C compiler for the host machine: cc (gcc 13.3.0 "cc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0")
[camera]   C linker for the host machine: cc ld.bfd 2.42
[camera]   C++ compiler for the host machine: c++ (gcc 13.3.0 "c++ (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0")
[camera]   C++ linker for the host machine: c++ ld.bfd 2.42
[camera]   Host machine cpu family: aarch64
[camera]   Host machine cpu: aarch64
[camera]   Header "fcntl.h" has symbol "F_ADD_SEALS" : YES 
[camera]   Header "unistd.h" has symbol "issetugid" : NO 
[camera]   Header "locale.h" has symbol "locale_t" : YES 
[camera]   Header "sys/mman.h" has symbol "memfd_create" : YES 
[camera]   Header "stdlib.h" has symbol "secure_getenv" : YES 
[camera]   Compiler for C supports arguments -Wno-c99-designator: NO 
[camera]   Found pkg-config: YES (/usr/bin/pkg-config) 1.8.1
[camera]   Found CMake: /usr/bin/cmake (3.28.3)
[camera]   Run-time dependency lttng-ust found: NO (tried pkgconfig and cmake)
[camera]   Program ./parser.py found: YES (/libcamera/utils/codegen/ipc/./parser.py)
[camera]   Program ./generate.py found: YES (/libcamera/utils/codegen/ipc/./generate.py)
[camera]   Program ./extract-docs.py found: YES (/libcamera/utils/codegen/ipc/./extract-docs.py)
[camera]   Configuring version.h using configuration
[camera]   Program openssl found: YES (/usr/bin/openssl)
[camera]   Run-time dependency libyuv found: NO (tried pkgconfig and cmake)
[camera]   Has header "libyuv.h" : NO 
[camera]   Library atomic found: YES
[camera]   Run-time dependency threads found: YES
[camera]   Run-time dependency libdw found: YES 0.190
[camera]   Run-time dependency libunwind found: YES 1.6.2
[camera]   Header "execinfo.h" has symbol "backtrace" : YES 
[camera]   Library rt found: YES
[camera]   Run-time dependency libpisp found: NO (tried pkgconfig and cmake)
[camera]   Looking for a fallback subproject for the dependency libpisp
[camera]   Executing subproject libpisp 
[camera]   libpisp| DEPRECATION: Option 'v4l2' value 'true' is replaced by 'enabled'
[camera]   libpisp| Project name: libpisp
[camera]   libpisp| Project version: 1.2.0
[camera]   libpisp| C compiler for the host machine: cc (gcc 13.3.0 "cc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0")
[camera]   libpisp| C linker for the host machine: cc ld.bfd 2.42
[camera]   libpisp| C++ compiler for the host machine: c++ (gcc 13.3.0 "c++ (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0")
[camera]   libpisp| C++ linker for the host machine: c++ ld.bfd 2.42
[camera]   libpisp| Configuring pisp_build_config.h using configuration
[camera]   libpisp| Run-time dependency nlohmann_json found: NO (tried pkgconfig and cmake)
[camera]   libpisp| Looking for a fallback subproject for the dependency nlohmann_json
[camera]   libpisp| Using subprojects/libpisp/subprojects/nlohmann_json.wrap
[camera]   libpisp| Downloading nlohmann_json source from https://github.com/nlohmann/json/releases/download/v3.11.2/include.zip
[camera]   Cloning into 'libyaml'...
[camera]   
[camera]   HEAD is now at 2c891fc Changes for v0.2.5 release
[camera]   
[camera]   Executing subproject libpisp:nlohmann_json
[camera]   nlohmann_json| DEPRECATION: Option 'v4l2' value 'true' is replaced by 'enabled'
[camera]   nlohmann_json| Project name: nlohmann_json
[camera]   nlohmann_json| Project version: 3.11.2
[camera]   nlohmann_json| C++ compiler for the host machine: c++ (gcc 13.3.0 "c++ (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0")
[camera]   nlohmann_json| C++ linker for the host machine: c++ ld.bfd 2.42
[camera]   nlohmann_json| Build targets in project: 25
[camera]   nlohmann_json| Subproject nlohmann_json finished.
[... 239 lines cut ...]
[camera]   [55/254] Compiling C++ object src/libcamera/base/libcamera-base.so.0.5.0.p/object.cpp.o
[camera]   [56/254] Compiling C++ object subprojects/libpisp/src/libpisp.so.1.2.0.p/helpers_v4l2_device.cpp.o
[camera]   [57/254] Compiling C++ object src/libcamera/base/libcamera-base.so.0.5.0.p/event_dispatcher_poll.cpp.o
[camera]   [58/254] Compiling C++ object src/libcamera/base/libcamera-base.so.0.5.0.p/thread.cpp.o
[camera]   [59/254] Compiling C object subprojects/libyaml/libyaml.a.p/src_emitter.c.o
[camera]   [60/254] Compiling C++ object subprojects/libpisp/src/libpisp.so.1.2.0.p/libpisp_backend_backend_prepare.cpp.o
[camera]   [61/254] Compiling C++ object subprojects/libpisp/src/libpisp.so.1.2.0.p/helpers_backend_device.cpp.o
[camera]   [62/254] Compiling C object subprojects/libyaml/libyaml.a.p/src_scanner.c.o
[camera]   [63/254] Compiling C++ object src/libcamera/base/libcamera-base.so.0.5.0.p/log.cpp.o
[camera]   [64/254] Compiling C++ object subprojects/libpisp/src/libpisp.so.1.2.0.p/libpisp_common_pisp_pwl.cpp.o
[camera]   [65/254] Compiling C++ object subprojects/libpisp/src/libpisp.so.1.2.0.p/helpers_media_device.cpp.o
[camera]   [66/254] Compiling C++ object subprojects/libpisp/src/libpisp.so.1.2.0.p/libpisp_backend_backend_default_config.cpp.o
[camera]   [67/254] Compiling C++ object subprojects/libpisp/src/libpisp.so.1.2.0.p/libpisp_backend_backend_debug.cpp.o
[camera]   ninja: build stopped: subcommand failed.
[camera]   Removing intermediate container 5757fdd6a9fc
[camera]   The command '/bin/sh -c cd libcamera &&     git submodule update --init &&     meson setup build --buildtype=release -Dpipelines=rpi/vc4,rpi/pisp -Dipas=rpi/vc4,rpi/pisp -Dv4l2=true -Dgstreamer=enabled -Dtest=false -Dlc-compliance=disabled -Dcam=disabled -Dqcam=disabled -Ddocumentation=disabled -Dpycamera=enabled &&     cd build &&     ninja &&     ninja install' returned a non-zero code: 1
[Info]     Uploading images
[Success]  Successfully uploaded images
[Error]    Some services failed to build:
[Error]      Service: camera
[Error]        Error: The command '/bin/sh -c cd libcamera &&     git submodule update --init &&     meson setup build --buildtype=release -Dpipelines=rpi/vc4,rpi/pisp -Dipas=rpi/vc4,rpi/pisp -Dv4l2=true -Dgstreamer=enabled -Dtest=false -Dlc-compliance=disabled -Dcam=disabled -Dqcam=disabled -Ddocumentation=disabled -Dpycamera=enabled &&     cd build &&     ninja &&     ninja install' returned a non-zero code: 1
[Info]     Built on c617b98
[Error]    Not deploying release.
Remote build failed
```

meawoppl, 2025-05-18:
There is something in meson.build that overwrited the PYTHONPATH variable before running the generation scripts. I don't understand meson well enough to triage, but if I add the needed path back in in `utils/codgen/build.meson` things work again.

naushir, 2025-05-19:
``PYTHONPATH`` gets set in the meson.build file here:
https://github.com/raspberrypi/libcamera/blob/d83ff0a4ae4503bc56b7ed48cd142c3dd423ad3b/utils/codegen/meson.build#L8

But the same build config works for me, so I don't think it's the direct cause of the failure.

@kbingham any thoughts about this build error?  ``jinja2`` seems to have been located by meson, but the gen-controls script fails to import the module.

naushir, 2025-05-19:
@meawoppl are you building libcamera from within a virtual environment?

kbingham, 2025-05-19:
@meawoppl Can you explain what you mean by  "Add the needed path back" means ? Are you manually setting it entirely? Are you adding something that was otherwise set by your more global system environment, or are you adding back in a path that libcamera needs to function ? 

I see this line in the debug prints:

```
[camera]   env PYTHONDONTWRITEBYTECODE=1 \
PYTHONPATH=/libcamera/utils/codegen /libcamera/utils/codegen/ipc/./parser.py \
 --output-root /libcamera/build \
--input-root /libcamera \
--mojoms ../include/libcamera/ipa/core.mojom
```

What's `env PYTHONDONTWRITEBYTECODE=1` coming from?

I wonder if for some reason meson is not getting the PYTHONPATH - so when it's performing the prepend - we only end up with `PYTHONPATH=/libcamera/utils/codegen` ?

naushir, 2025-05-19:
> What's `env PYTHONDONTWRITEBYTECODE=1` coming from?

This comes from the meson build file:
https://github.com/raspberrypi/libcamera/blob/d83ff0a4ae4503bc56b7ed48cd142c3dd423ad3b/utils/codegen/meson.build#L7

kbingham, 2025-05-19:
Aha, So it looks like that script is indeed completely ignoring any externally set PYTHONPATH ...

```
py_build_env = environment()
# \todo Investigate usage of PYTHONPYCACHEPREFIX for Python >= 3.8
py_build_env.set('PYTHONDONTWRITEBYTECODE', '1')
py_build_env.prepend('PYTHONPATH', meson.current_source_dir())
```

Looks like it will only keep those variables in the path - so if @meawoppl's environment is relying on any externally / non-system paths being specified then it wouldn't be accounted for here ?

We certainly need to know more about the build environment to understand or reproduce this ...

naushir, 2025-05-23:
@meawoppl any updates on this issue?