raspberrypi/libcamera #267: No video stream on buildroot

elcalzado, 2025-05-26:
I'm trying to use a Raspberry Pi Zero 2 W with a Camera Module 3 as a USB webcam. I get the [uvc-gadget](https://gitlab.freedesktop.org/camera/uvc-gadget) to detect my camera and show up on the host computer but it won't stream video. I can get the cam app to record frames and I was able to convert them to png using ffmpeg. If anyone could help me decipher what is missing for the video to properly stream that would be a huge help.

On buildroot I selected libcamera and all of the options under it.

**This is what it looks like when I run the uvc-gadget:**
```
Loading composite module
[0:04:30.109430886] [245] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_vc4.so: IPA module /usr/lib/libcamera/ipa_rpi_vc4.so is signed
[0:04:30.110428282] [245] DEBUG IPAManager ipa_manager.cpp:229 Loaded IPA module '/usr/lib/libcamera/ipa_rpi_vc4.so'
[0:04:30.110766772] [245]  INFO Camera camera_manager.cpp:326 libcamera v0.5.0
[0:04:30.113546563] [249] DEBUG Camera camera_manager.cpp:73 Starting camera manager
[0:04:30.114054376] [249] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "unicam" created from /dev/media2
[0:04:30.114955522] [249] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: unicam
[0:04:30.115193074] [249] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "bcm2835-isp" created from /dev/media0
[0:04:30.115769219] [249] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: bcm2835-isp
[0:04:30.115973282] [249] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "bcm2835-isp" created from /dev/media1
[0:04:30.116588438] [249] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: bcm2835-isp
[0:04:30.116699792] [249] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'rpi/vc4'
[0:04:30.116790990] [249] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "unicam"
[0:04:30.116892553] [249] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "bcm2835-isp"
[0:04:30.117080105] [249] DEBUG DmaBufAllocator dma_buf_allocator.cpp:112 Using /dev/dma_heap/linux,cma
[0:04:30.117231251] [249] DEBUG CameraSensor camera_sensor_raw.cpp:210 imx708: unsupported number of sinks (0) or sources (2)
[0:04:30.117423074] [249] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Exposure (0x00980911)
[0:04:30.117549480] [249] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Horizontal Flip (0x00980914)
[0:04:30.117650365] [249] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Vertical Flip (0x00980915)
[0:04:30.117754480] [249] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Wide Dynamic Range (0x009a0915)
[0:04:30.117867344] [249] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Camera Orientation (0x009a0922)
[0:04:30.117992605] [249] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Camera Sensor Rotation (0x009a0923)
[0:04:30.118098959] [249] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Vertical Blanking (0x009e0901)
[0:04:30.118197344] [249] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Horizontal Blanking (0x009e0902)
[0:04:30.118295886] [249] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Analogue Gain (0x009e0903)
[0:04:30.118393959] [249] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Red Pixel Value (0x009e0904)
[0:04:30.118503751] [249] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Green (Red) Pixel Value (0x009e0905)
[0:04:30.118603699] [249] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Blue Pixel Value (0x009e0906)
[0:04:30.118701876] [249] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Green (Blue) Pixel Value (0x009e0907)
[0:04:30.118798803] [249] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Notify Gains (0x009e0909)
[0:04:30.118908230] [249] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Link Frequency (0x009f0901)
[0:04:30.119014219] [249] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Pixel Rate (0x009f0902)
[0:04:30.453274219] [249] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Test Pattern (0x009f0903)
[0:04:30.453422553] [249] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Digital Gain (0x009f0905)
[0:04:30.535399428] [249] DEBUG V4L2 v4l2_device.cpp:721 'dw9807 10-000c': Control: Focus, Absolute (0x009a090a)
[0:04:30.536926928] [249] DEBUG CameraSensor camera_sensor_legacy.cpp:1023 'imx708': Apply test pattern mode 0
[0:04:30.537083126] [249] DEBUG CameraSensor camera_sensor.cpp:468 Entity 'imx708' matched by CameraSensorLegacy
[0:04:30.564205782] [249] DEBUG IPAManager ipa_manager.cpp:300 IPA module /usr/lib/libcamera/ipa_rpi_vc4.so signature is not valid
[0:04:30.564651042] [249] DEBUG IPAProxy raspberrypi_ipa_proxy.cpp:45 initializing raspberrypi proxy: loading IPA from /usr/lib/libcamera/ipa_rpi_vc4.so
[0:04:30.575331615] [249] DEBUG Timer timer.cpp:93 Starting timer 0x7f8ee83d40: deadline 0:04:32.575297396
[0:04:30.575543126] [249] DEBUG Event event_dispatcher_poll.cpp:215 next timer 0x7f8ee83d40 expires in 1.999768385
[0:04:30.637085886] [249] DEBUG RPiDelayedControls delayed_controls.cpp:103 Set a delay of 3 and priority write flag 1 for Vertical Blanking
[0:04:30.637246459] [249] DEBUG RPiDelayedControls delayed_controls.cpp:103 Set a delay of 3 and priority write flag 0 for Horizontal Blanking
[0:04:30.637326198] [249] DEBUG RPiDelayedControls delayed_controls.cpp:103 Set a delay of 2 and priority write flag 0 for Exposure
[0:04:30.637402865] [249] DEBUG RPiDelayedControls delayed_controls.cpp:103 Set a delay of 2 and priority write flag 0 for Analogue Gain
[0:04:30.637867865] [249] DEBUG V4L2 v4l2_videodevice.cpp:631 /dev/video1[20:cap]: Opened device platform:3f801000.csi: unicam: unicam
[0:04:30.638326667] [249] DEBUG V4L2 v4l2_videodevice.cpp:631 /dev/video2[21:cap]: Opened device platform:3f801000.csi: unicam: unicam
[0:04:30.638492396] [249] DEBUG V4L2 v4l2_device.cpp:721 /dev/video13[22:cap]: Control: Red Balance (0x0098090e)
[0:04:30.638613021] [249] DEBUG V4L2 v4l2_device.cpp:721 /dev/video13[22:cap]: Control: Blue Balance (0x0098090f)
[0:04:30.638718907] [249] DEBUG V4L2 v4l2_device.cpp:721 /dev/video13[22:cap]: Control: Colour Correction Matrix (0x009819e1)
[0:04:30.638822657] [249] DEBUG V4L2 v4l2_device.cpp:721 /dev/video13[22:cap]: Control: Lens Shading (0x009819e2)
[0:04:30.638923178] [249] DEBUG V4L2 v4l2_device.cpp:721 /dev/video13[22:cap]: Control: Black Level (0x009819e3)
[0:04:30.639023490] [249] DEBUG V4L2 v4l2_device.cpp:721 /dev/video13[22:cap]: Control: Green Equalisation (0x009819e4)
[0:04:30.639139480] [249] DEBUG V4L2 v4l2_device.cpp:721 /dev/video13[22:cap]: Control: Gamma (0x009819e5)
[0:04:30.639238803] [249] DEBUG V4L2 v4l2_device.cpp:721 /dev/video13[22:cap]: Control: Denoise (0x009819e6)
[0:04:30.639337709] [249] DEBUG V4L2 v4l2_device.cpp:721 /dev/video13[22:cap]: Control: Sharpen (0x009819e7)
[0:04:30.639438230] [249] DEBUG V4L2 v4l2_device.cpp:721 /dev/video13[22:cap]: Control: Defective Pixel Correction (0x009819e8)
[0:04:30.639539063] [249] DEBUG V4L2 v4l2_device.cpp:721 /dev/video13[22:cap]: Control: Colour Denoise (0x009819e9)
[0:04:30.639643750] [249] DEBUG V4L2 v4l2_device.cpp:721 /dev/video13[22:cap]: Control: Digital Gain (0x009f0905)
[0:04:30.639802344] [249] DEBUG V4L2 v4l2_videodevice.cpp:631 /dev/video13[22:out]: Opened device platform:bcm2835-isp: bcm2835-isp: bcm2835-isp
[0:04:30.640266563] [249] DEBUG V4L2 v4l2_videodevice.cpp:631 /dev/video14[23:cap]: Opened device platform:bcm2835-isp: bcm2835-isp: bcm2835-isp
[0:04:30.640506250] [249] DEBUG V4L2 v4l2_videodevice.cpp:631 /dev/video15[24:cap]: Opened device platform:bcm2835-isp: bcm2835-isp: bcm2835-isp
[0:04:30.640691928] [249] DEBUG V4L2 v4l2_videodevice.cpp:631 /dev/video16[25:cap]: Opened device platform:bcm2835-isp: bcm2835-isp: bcm2835-isp
[0:04:30.640854636] [249]  INFO RPI vc4.cpp:401 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media2 and ISP device /dev/media0
[... 247 lines cut ...]
[0:06:00.476140122] [265] DEBUG V4L2 v4l2_videodevice.cpp:1844 /dev/video1[20:cap]: Dequeuing buffer 0
[0:06:00.476296529] [265] DEBUG RPI vc4.cpp:772 Stream Unicam Image buffer dequeue, buffer id 1, timestamp: 360290528000
[0:06:00.476395904] [265] DEBUG RPiDelayedControls delayed_controls.cpp:220 Reading Exposure to 2171 at index 0
[0:06:00.476481633] [265] DEBUG RPiDelayedControls delayed_controls.cpp:220 Reading Horizontal Blanking to 3680 at index 0
[... cut ...]
[0:06:00.830353872] [265] DEBUG Request request.cpp:129 Request(1:X:0/1:1)
[0:06:00.830438351] [265] DEBUG Request request.cpp:129 Request(2:X:0/1:2)
[0:06:00.830519654] [265] DEBUG Request request.cpp:129 Request(3:X:0/1:3)
[0:06:00.830615904] [265] ERROR IPCUnixSocket ipc_unixsocket.cpp:192 Failed to send: Transport endpoint is not connected
[0:06:00.830705174] [265] ERROR IPCPipe ipc_pipe_unixsocket.cpp:66 Failed to call sync
[0:06:00.830773872] [265] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:317 Failed to call stop
```

elcalzado, 2025-05-27:
Pretty sure the error has to do with:
`[249] DEBUG IPAManager ipa_manager.cpp:300 IPA module /usr/lib/libcamera/ipa_rpi_vc4.so signature is not valid`

That message is not present in Raspberry Pi OS so I'll look into that but I'm not sure how to fix that.

kbingham, 2025-05-27:
Make sure you have the (optional) dependencies for module signing installed in buildroot both at compilation and runtime.

```
for IPA module signing: [recommended]

    Either libgnutls28-dev or libssl-dev, openssl

    Without IPA module signing, all IPA modules will be isolated in a separate process. This adds an unnecessary extra overhead at runtime.
```

kbingham, 2025-05-27:
It looks like the fault starts around here:

```
[0:03:56.078508295] [260] FATAL Serializer control_serializer.cpp:604 A list of V4L2 controls requires an ControlInfoMap
Backtrace:
/lib64/libcamera.so.0.5(_ZN9libcamera17ControlSerializer11deserializeINS_11ControlListEEET_RNS_16ByteStreamBufferE+0x300) [0x7f87848f10]
/lib64/libcamera.so.0.5(_ZN9libcamera17IPADataSerializerINS_11ControlListEE11deserializeEN9__gnu_cxx17__normal_iteratorIPKhSt6vectorIhSaIhEEEESA_PNS_17ControlSerializerE+0xfc) [0x7f878591bc]
/usr/libexec/libcamera/raspberrypi_ipa_proxy(+0x149c0) [0x555eb6c9c0]
/lib64/libcamera.so.0.5(_ZN9libcamera13IPCUnixSocket12dataNotifierEv+0x264) [0x7f878642d4]
/lib64/libcamera-base.so.0.5(_ZN9libcamera19EventDispatcherPoll16processNotifiersERKSt6vectorI6pollfdSaIS2_EE+0x1c4) [0x7f8772b5b4]
/lib64/libcamera-base.so.0.5(_ZN9libcamera19EventDispatcherPoll13processEventsEv+0x298) [0x7f8772bc18]
/usr/libexec/libcamera/raspberrypi_ipa_proxy(+0x95b8) [0x555eb615b8]
/lib64/libc.so.6(+0x2c32c) [0x7f8734d32c]
/lib64/libc.so.6(__libc_start_main+0x98) [0x7f8734d408]
/usr/libexec/libcamera/raspberrypi_ipa_proxy(+0x9ab0) [0x555eb61ab0]
```

kbingham, 2025-05-28:
There seems to be something specific here in this build. I can use both the RKISP and the RPi in isolated mode and it works too ;-)

Just confirmed on my Pi5 running

[0:04:19.615646580] [984]  INFO Camera camera_manager.cpp:326 libcamera v0.5.0+67-c9843cb9
[0:04:19.629073477] [985]  INFO RPI pisp.cpp:720 libpisp version v1.2.1 981977ff21f3 29-04-2025 (14:13:50)

with an IMX283 can capture fine with an Isolated IPA.

elcalzado, 2025-05-28:
Is there a common reason why an IPA module signature might come out as invalid? From what I understand, buildroot makes sure to sign the modules but I'm not sure how singing works and how it might work differently if cross-compiling. 

<details><summary>There's this section in the <code>libcamera.mk</code> for buildroot which might be involved</summary>
<pre><code># Open-Source IPA shlibs need to be signed in order to be runnable within the
# same process, otherwise they are deemed Closed-Source and run in another
# process and communicate over IPC.
# Buildroot sanitizes RPATH in a post build process. meson gets rid of rpath
# while installing so we don't need to do it manually here.
# Buildroot may strip symbols, so we need to do the same before signing
# otherwise the signature won't match the shlib on the rootfs. Since meson
# install target is signing the shlibs, we need to strip them before.
LIBCAMERA_STRIP_FIND_CMD = \
	find $(@D)/build/src/ipa \
	$(if $(call qstrip,$(BR2_STRIP_EXCLUDE_FILES)), \
		-not \( $(call findfileclauses,$(call qstrip,$(BR2_STRIP_EXCLUDE_FILES))) \) ) \
	-type f -name 'ipa_*.so' -print0

define LIBCAMERA_BUILD_STRIP_IPA_SO
	$(LIBCAMERA_STRIP_FIND_CMD) | xargs --no-run-if-empty -0 $(STRIPCMD)
endef

LIBCAMERA_POST_BUILD_HOOKS += LIBCAMERA_BUILD_STRIP_IPA_SO</code></pre></details>

I'll attach the .config for packages, the .config for the kernel and my sdcard.img if anyone wants to try and replicate my environment. 

As of now you have to modprobe bcm2835_isp manually before running uvc-gadget.sh.

[buildroot_env.zip](https://www.mediafire.com/file/qs65kqd9k68x2vu/buildroot_env.zip/file)

elcalzado, 2025-05-29:
Issue #50 seems related to what I'm struggling with but I'm unable to fully review or test it at this moment I'll update when I get a chance to look at it. I'm particularly interested in the mention of stripping the signed binaries.

kbingham, 2025-05-29:
If the binaries (IPAs) are being modified after signing by buildroot - then that would break the signatures yes.
First obvious check - is the .sign file in the same location as the ipa ?

Note there are two separate issues here:

 - 1) The modules are not signed correctly - or potentially not verified successfully
 - 2) When isolated - the IPA currently hits a crash regarding control handling. 
 
Both of those issues are separate, and solving either one of them may be sufficient to get your device working, if the control handling bug does not occur when the modules are not isolated.

Note - now I wrote that - I realise you are testing on a pizero - while I tested isolated modules on the Pi5 ... there could certainly be differences there between pi0 and pi5 ... so somehow we should also test isolated IPA on pizero running native Raspberry Pi OS to see if there are any hidden bugs there.

elcalzado, 2025-05-30:
> Note - now I wrote that - I realise you are testing on a pizero - while I tested isolated modules on the Pi5 ... there could certainly be differences there between pi0 and pi5 ... so somehow we [...]

I tested the modules in isolation (rpi/vc4) on Raspberry Pi OS Lite (Bookworm, 64 bit) using Raspberry Pi Zero 2 W and IMX708. The result was exactly the same as on buildroot. Same erros and behavior.

I used the newer buildroot commit you mentioned and the module signature is now valid! Unfortunately still no video stream and the `configfs-gadget.g1 gadget.0: uvc: VS request completed with status -61` message persists.

<details><summary>New <code>LIBCAMERA_LOG_LEVELS=*:0 uvc-gadget.sh</code> output</summary>
<pre><code>Loading composite module
Detecting platform:
  board : Raspberry Pi Zero 2 W Rev 1.0
  udc   : 3f980000.usb
Creating the USB gadget
Creating gadget directory g1
OK
Setting Vendor and Product ID's
OK
Setting English strings
OK
Creating Config
Creating functions...
  Creating UVC gadget functionality : uvc.0
Renaming function_name
OK
Binding USB D[   30.448457] configfs-gadget.g1 gadget.0: uvc: uvc_function_bind()
evice Controller[   30.456178] dwc2 3f980000.usb: bound driver configfs-gadget.g1

OK
[0:00:31.018247227] [251] DEBUG IPAModule ipa_module.cpp:333 ipa_mali_c55.so: IPA module /usr/lib/libcamera/ipa_mali_c55.so is signed
[0:00:31.019935352] [251] DEBUG IPAManager ipa_manager.cpp:229 Loaded IPA module '/usr/lib/libcamera/ipa_mali_c55.so'
[0:00:31.036439936] [251] DEBUG IPAModule ipa_module.cpp:333 ipa_rkisp1.so: IPA module /usr/lib/libcamera/ipa_rkisp1.so is signed
[0:00:31.036675144] [251] DEBUG IPAManager ipa_manager.cpp:229 Loaded IPA module '/usr/lib/libcamera/ipa_rkisp1.so'
[0:00:31.057523686] [251] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_vc4.so: IPA module /usr/lib/libcamera/ipa_rpi_vc4.so is signed
[0:00:31.057758321] [251] DEBUG IPAManager ipa_manager.cpp:229 Loaded IPA module '/usr/lib/libcamera/ipa_rpi_vc4.so'
[0:00:31.066815977] [251] DEBUG IPAModule ipa_module.cpp:333 ipa_soft_simple.so: IPA module /usr/lib/libcamera/ipa_soft_simple.so is signed
[0:00:31.067019779] [251] DEBUG IPAManager ipa_manager.cpp:229 Loaded IPA module '/usr/lib/libcamera/ipa_soft_simple.so'
[0:00:31.069241654] [251] DEBUG IPAModule ipa_module.cpp:333 ipa_vimc.so: IPA module /usr/lib/libcamera/ipa_vimc.so is signed
[0:00:31.069395665] [251] DEBUG IPAManager ipa_manager.cpp:229 Loaded IPA module '/usr/lib/libcamera/ipa_vimc.so'
[0:00:31.069472800] [251]  INFO Camera camera_manager.cpp:326 libcamera v0.5.0
[0:00:31.069863217] [255] DEBUG Camera camera_manager.cpp:73 Starting camera manager
[0:00:31.070195092] [255] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "bcm2835-isp" created from /dev/media2
[0:00:31.072273009] [255] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: bcm2835-isp
[0:00:31.072460144] [255] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "unicam" created from /dev/media0
[0:00:31.072914415] [255] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: unicam
[0:00:31.073206081] [255] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "bcm2835-codec" created from /dev/media3
[0:00:31.074015144] [255] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: bcm2835-codec
[0:00:31.074228061] [255] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "bcm2835-isp" created from /dev/media1
[0:00:31.074631081] [255] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: bcm2835-isp
[0:00:31.074707384] [255] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'imx8-isi'
[0:00:31.074779102] [255] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'mali-c55'
[0:00:31.074864519] [255] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'rkisp1'
[0:00:31.080287800] [255] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'rpi/vc4'
[0:00:31.082855300] [255] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "unicam"
[0:00:31.082962592] [255] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "bcm2835-isp"
[0:00:31.083091238] [255] DEBUG ClockRec clock_recovery.cpp:69 configure 100 2000 10 50000
[0:00:31.083141706] [255] DEBUG ClockRec clock_recovery.cpp:83 reset
[0:00:31.083227019] [255] DEBUG DmaBufAllocator dma_buf_allocator.cpp:112 Using /dev/dma_heap/linux,cma
[0:00:31.083319884] [255] DEBUG CameraSensor camera_sensor_raw.cpp:210 imx708: unsupported number of sinks (0) or sources (2)
[0:00:31.086052904] [255] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Exposure (0x00980911)
[0:00:31.086196706] [255] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Horizontal Flip (0x00980914)
[0:00:31.086258686] [255] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Vertical Flip (0x00980915)
[0:00:31.086322331] [255] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Wide Dynamic Range (0x009a0915)
[0:00:31.086382436] [255] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Camera Orientation (0x009a0922)
[0:00:31.420586394] [255] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Camera Sensor Rotation (0x009a0923)
[0:00:31.420664727] [255] DEBUG V4L2 v4l2_device.cpp:721 'imx708': Control: Vertical Blanking (0x009e0901)
[... cut ...]
[0:00:38.861293526] [255] ERROR RPI pipeline_base.cpp:1358 Camera frontend has timed out!
[0:00:38.861337016] [255] ERROR RPI pipeline_base.cpp:1359 Please check that your camera sensor connector is attached securely.
[0:00:38.861342276] [263] DEBUG IPARPI cam_helper_imx708.cpp:111 Embedded buffer size: 28800
[0:00:38.861395453] [255] ERROR RPI pipeline_base.cpp:1360 Alternatively, try another cable and/or sensor.
```

I again tested on Pi OS without moving anything to make sure that no hardware issues were present and it worked fine.

elcalzado, 2025-05-31:
I compared the logs for buildroot and pi os and so far can't point out anything obvious. The only thing I noticed is that at line 271 of the pi os log this message shows up earlier than in the buildroot log:

```
[0:04:48.560546951] [671] DEBUG RPI pipeline_base.cpp:1378 Frame start 0
[0:04:48.560593824] [671] DEBUG RPiDelayedControls delayed_controls.cpp:240 frame 0 started
```

I also noticed that for buildroot the ClockRec messages always share the same number. For example:

```
[0:01:43.090722096] [264] DEBUG ClockRec clock_recovery.cpp:141 addSample 103090718 103090718
```

On pi os the second number changes:

```
[0:04:48.429330029] [671] DEBUG ClockRec clock_recovery.cpp:141 addSample 288429325 1748650079981446
```

Additionally I found these two new error messges in buildroot:

```
[0:01:48.272200739] [264] ERROR V4L2 v4l2_device.cpp:390 'dw9807 10-000c': Unable to set controls: Connection timed out
# More lines here
[0:01:48.453997458] [264] ERROR V4L2 v4l2_device.cpp:390 'dw9807 10-000c': Unable to set controls: Input/output error
```

[buildroot_libcamera.log](https://github.com/user-attachments/files/20528401/buildroot_libcamera.log)
[pioslite_libcamera.log](https://github.com/user-attachments/files/20528402/pioslite_libcamera.log)

elcalzado, 2025-06-05:
I've been trying to work on this and figure out what causes the issue but so far haven't been able to figure it out. Either way, though it'd be useful to update with some things I've tried and share my current image file.

Ran a few tests to try and see if anything obvious was missing, so far nothing I can point out.

[buildroot_ldd.log](https://github.com/user-attachments/files/20601540/buildroot_ldd.log)
[buildroot_lddtree.log](https://github.com/user-attachments/files/20601539/buildroot_lddtree.log)
[buildroot_trace.log](https://github.com/user-attachments/files/20601538/buildroot_trace.log)

[pioslite_ldd.log](https://github.com/user-attachments/files/20601542/pioslite_ldd.log)
[pioslite_lddtree.log](https://github.com/user-attachments/files/20601543/pioslite_lddtree.log)
[pioslite_trace.log](https://github.com/user-attachments/files/20601544/pioslite_trace.log)

<details><summary>I also get this <code>dmesg</code> error when running the uvc-gadget and trying to get video to OBS</summary>
<pre><code>[  176.254916] ------------[ cut here ]------------
[  176.264156] Firmware transaction 0x00030046 timeout
[  176.279576] WARNING: CPU: 1 PID: 26 at drivers/firmware/raspberrypi.c:131 rpi_firmware_property_list+0x238/0x290
[  176.291973] Modules linked in:
[  176.307267]  usb_f_uvc
[  176.325079]  uvc
[  176.335770]  videobuf2_dma_sg
[  176.345767]  libcomposite
[  176.355244]  brcmfmac
[  176.365848]  brcmutil
[  176.376106]  cfg80211
[  176.386016]  raspberrypi_hwmon
[  176.395926]  imx708
[  176.405837]  dw9807_vcm
[  176.416529]  dwc2
[  176.426265]  i2c_bcm2835
[  176.436349]  i2c_mux_pinctrl
[  176.445912]  i2c_mux
[  176.456083]  vc4
[  176.466601]  snd_soc_hdmi_codec
[  176.476425]  drm_display_helper
[  176.485901]  cec
[  176.496679]  drm_dma_helper
[  176.507458]  drm_kms_helper
[  176.516934]  drm
[  176.527365]  drm_panel_orientation_quirks
[  176.537797]  snd_soc_core
[  176.547273]  snd_compress
[  176.558920]  snd_pcm_dmaengine
[  176.569177]  snd_pcm
[  176.579435]  snd_timer
[  176.590127]  snd
[  176.599950]  backlight
[  176.609947]  raspberrypi_gpiomem
[  176.619423]  bcm2835_unicam_legacy
[  176.629420]  hci_uart
[  176.640286]  btbcm
[  176.651325]  bluetooth
[  176.661235]  ecdh_generic
[  176.670885]  ecc
[  176.680882]  rfkill
[  176.691140]  libaes
[  176.700616]  uio_pdrv_genirq
[  176.710352]  uio
[  176.720089]
[  176.730611] CPU: 1 UID: 0 PID: 26 Comm: kworker/1:0 Not tainted 6.12.25-v8 #2
[  176.740084] Hardware name: Raspberry Pi Zero 2 W Rev 1.0 (DT)
[  176.749215] Workqueue: events get_values_poll [raspberrypi_hwmon]
[  176.763985]
[  176.777370] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  176.791099] pc : rpi_firmware_property_list+0x238/0x290
[  176.800228] lr : rpi_firmware_property_list+0x238/0x290
[  176.814827] sp : ffffffc080103ca0
[  176.827689] x29: ffffffc080103cc0
[  176.840551]  x28: 0000000000000000
[  176.851503]  x27: ffffff8002b00840
[  176.862454]
[... 53 lines cut ...]
drm_kms_helper        225280  2 drm_dma_helper,vc4
drm                   671744  5 drm_kms_helper,drm_dma_helper,vc4,drm_display_helper
drm_panel_orientation_quirks    28672  1 drm
snd_soc_core          303104  2 vc4,snd_soc_hdmi_codec
snd_compress           20480  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_pcm               147456  4 snd_soc_hdmi_codec,snd_compress,snd_soc_core,snd_pcm_dmaengine
snd_timer              36864  1 snd_pcm
snd                   114688  5 snd_soc_hdmi_codec,snd_timer,snd_compress,snd_soc_core,snd_pcm
backlight              24576  2 drm_kms_helper,drm
raspberrypi_gpiomem    12288  0
bcm2835_unicam_legacy    53248  5
hci_uart               53248  0
btbcm                  24576  1 hci_uart
bluetooth             647168  3 hci_uart,btbcm
ecdh_generic           16384  1 bluetooth
ecc                    40960  1 ecdh_generic
rfkill                 32768  2 bluetooth,cfg80211
libaes                 12288  1 bluetooth
uio_pdrv_genirq        12288  0
uio                    20480  1 uio_pdrv_genirq</code></pre></details>

The new buildroot image should automatically connect to the host pc and appear as a webcam device. It still wont stream but everything else works. If anyone tries it and wants to get more debug output from libcamera remove start-uvc-gadget.sh from startup in the inittab, reboot, and run it manually using LIBCAMERA_LOG_LEVELS.

In the [buildroot_env.zip](https://www.mediafire.com/file/skbeomxycmldiyn/buildroot_env.zip/file) im also including the package/libcamera and package/uvc-gadget dirs, .config for packages and .config for linux kernel, and the sdcard.img. It's all setup for the Raspberry Pi Zero 2 W and Camera Module 3.