raspberrypi/libcamera #275: RPi HD Camera "hangs up" following upgrade from libcamera 0.3y to 0.5y -- Please Help!

ldb20151, 2025-06-17:
[RPi Camera Lockup with libcamera 0.5.pdf](https://github.com/user-attachments/files/20785142/RPi.Camera.Lockup.with.libcamera.0.5.pdf)

I'm a newby; I need expert help to figure out how to diagnose/fix this problem.  I've tried to resolve this on my own but am not making progress.  I'm stuck!

I've been working on an Indi-AllSky camera system for over a year.  As I was nearly complete, I hit this camera hangup issue and have been unable to get past it.  I'm reaching out here in the hope someone will recognize the issue or at least be able to steer me closer to the answer I need.  I'm pretty far over my head in this area, but I'm slowly learning.  If I say something stupid, please consider the source :)

**The Problem**
With the latest version of licamera, my Raspberry Pi HD Camera "hangs" soon after launching the Indi-AllSky package (Git Hub, Developer Aaron Morris).  The Indi-AllSky developer suggests the issue is likely with libcamera.  To check, I have tried two new HD cameras and three different camera cables but these make no difference.  All works great on 0.3y, but hangs up after the upgrade to 0.5y.  Its very repeatable.  Though the issue seems unrelated to hardware, I easily replicate the issue by can swapping between upgraded/non-upgraded software versions installed on different NMVe SSDs.  

Below is an example of the error message I receive and the note on this topic I received from the Indi-AllSky developer.  I'm also attaching the system support details Aaron refers to below using both libcamera 0.3y and libcamera 0.5y.  I highlighted what I believe is the relevant info but am including all because I'm unsure what might be useful.  I would leave the system in this configuration but I need other package upgrades in place for security and functionality the newer version brings. 

v/r,
Larry B.
 
[image]

**Message From Indi-AllSky developer Aaron Morris on GitHub**
"_aaronwmorris
on May 17
Maintainer
Have you compared the versions of libcamera between the working system and the system that hangs? There is nothing in indi-allsky that would cause the rpicam-still program to hang.
A Support Info for the old and new systems would show this detail._"

EricClaeys, 2025-06-19:
@naushir,
my "Allsky" and Aaron's "indi allsky" are similar in terms of functionality, but server slightly different audiences.
I think since both indi allsky and Allsky users are having the same problem, it's very unlikely the hanging is due to our programs.
@MrNbaphoto and 
~~~
rpicam-hello --version
rpicam-apps build: v1.6.0 https://github.com/raspberrypi/rpicam-apps/commit/025ca84648c9b9d74711477bf94b05bec349f53d 03-02-2025 (16:21:04)
rpicam-apps capabilites: egl:1 qt:1 drm:1 libav:1
libcamera build: v0.4.0+53-29156679
~~~

<hr>

@ldb20151,
For consistency with what other people have run to show the problem, may I suggest running the following:
~~~
while true; do
    rpicam-still --thumb none --output 'test.jpg' --timeout 1 --nopreview --width 4056 --height 3040 --shutter 5ms --analoggain 1 --awb auto --quality 100 > /tmp/test.txt 2>&1
done
~~~
When `rpicam-still` hangs it will not return, so your check for `if [ $? -ne 0]` will never be run.  Also, the above code will only produce one .jpg file; your code will produce hundreds or thousands, depending on how long it takes rpicam-still to hang.

Eric

ldb20151, 2025-06-20:
I’ll give it a shot. I’m currently  burning- in a new version of allsky to see if the signature changed. That’ll finish late tomorrow. 
I appreciate the suggested code. I definitely don’t want thousands of test files. Thx!  
However, I’ve noticed a strong correlation between hangup frequency of occurrence and exposure duration. Is there a way to drive rpicam-still to take long exposures?  Might be more likely to catch something. 
Larry

naushir, 2025-06-20:
> However, I’ve noticed a strong correlation between hangup frequency of occurrence and exposure duration. Is there a way to drive rpicam-still to take long exposures? Might be more likely to catch something. Larry

You can use the following command line parameters to setup a long exposure:
```
rpicam-still --framerate 0 --shutter 10s ...
```

and change the ``--shutter`` value as appropriate for a larger shutter speed.

ldb20151, 2025-06-20:
Im running a pi5 8gb with pi HD camera.
I hope to run the above test this weekend.
-Larry

EricClaeys, 2025-06-20:
@naushir,
What is the advantage of using `--framerate 0`?  Should Allsky use it?

Eric

naushir, 2025-06-20:
> [@naushir](https://github.com/naushir), What is the advantage of using `--framerate 0`? Should Allsky use it?
> 

Actually I was questioning that myself.  In ``rpicam-vid`` if you ask for an exposure time > 33ms, it will clip since the default framerate is 30.  So you don't specify a framerate limit `--framerate 0` in such cases.

I don't think it works the same for ``rpicam-still`` and is probably not needed there.

ldb20151, 2025-06-20:
[@naushir](https://github.com/naushir),

I successfully ran the attached camera stress test file.  I ran this just 1hr at a time, 8 times.  All successful.  The test loop is very slightly modified from the suggested file to do just 1hr sessions as I was concerned about the size of the files. A sample test log file and the exact loop I used to test are attached.  It seemed to function as expected -- hope my tweaks didn't defeat the purpose.  

Normally, the hangup repeats several times within an hour, so I though that should catch the problem if its there.  Sadly I guess not.

At this point I'm unsure what to think.  Since the latest update to Indi-AllSky fixed an issue that the dev thought may have caused my problem, I think I've had only 1 hangup.  That one was at max allowable exposure (120sec).  Since I backed down a bit, no hangups.  Time will tell.  I am still seeing different but still puzzling & rather frequent messages as below, but I don't see any impact.  I acknowledge the message and the system presses forward without issue.  

[image]

v/r,
-larry

[test.pdf](https://github.com/user-attachments/files/20843486/test.pdf)

[Stress Test Capture_Loop June 2025.pdf](https://github.com/user-attachments/files/20843508/Stress.Test.Capture_Loop.June.2025.pdf)

EricClaeys, 2025-06-21:
@ldb20151,
If you replace the `>>` with `>` in this line you don't need to worry about file size:
~~~
echo "[$(date)] Capturing image..." >> "$logfile" 
~~~
since it will recreate the file for every image.

ldb20151, 2025-06-22:
That's helpful, thank you!

naushir, 2025-06-23:
> At this point I'm unsure what to think. Since the latest update to Indi-AllSky fixed an issue that the dev thought may have caused my problem, I think I've had only 1 hangup.

Do you happen to know what this issue was in Indi-Allsky that was fixed?  It might shed some light on the other issue.

EricClaeys, 2025-07-11:
@naushir,
I assume you fixed it on your end since you closed the Issue?
How do users get the fix?

ldb20151, 2025-07-12:
@EricClaeys ,

I’m uncertain but I believe it’s been closed because the lockup I was experiencing (very frequently) was traced to an issue with Indi-AllSky application that I was using. Unfortunately I do not know what the indi-AllSky dev did on his end to resolve the issue. 

As mentioned above, since the allsky fix, I’ve only had few (3-4) lockups. All occurred using the same allsky app but the lockup doesn’t have a repeatable signature. Seems very random for me at the moment. 

So for myself I’m going to wait and watch to see if I can figure out how to repeat the lockup. Without a reproduceable lockup signature it’s almost impossible to find root cause. I know that’s not helpful. I don’t know any other way to do it.

EricClaeys, 2025-07-12:
@ldb20151,
I'll ask Aaron what he changed in indi allsky to work around the bug.  In Allsky we now timeout the libcamera-still command after a few minutes as a workaround.

In my view, when a program like libcamera-still is called and a lockup occurs in the operating system, that's a bug, either in libcamera-still or in the operating system.  If indi allsky or Allsky are calling libcamera-still "incorrectly," libcamera-still should produce an error message, not hang.
I realize it's difficult for developers to find the cause when they cannot reproduce the problem, and that the 500+ RPi cameras indi allsky and Allsky represent is probably a drop in the bucket for the Raspberry company, so at some point they are likely to say "we give up".  When that happens I would hope they would say that rather than just close the Issue with no reason.

naushir, 2025-07-12:
I incorrectly read the earlier comments and thought the issue was resolved by changes in the Indi-AllSky application.  Since that does not seem to be the case, I'll re-open this.

I should mention again, I've been trying to reproduce this and https://github.com/raspberrypi/rpicam-apps/issues/799 for several weeks now without success.  This is across several boards running captures in loops for days.  So far, I've not this crash, so there is really nothing I can suggest right now.

EricClaeys, 2025-07-12:
@naushir, thanks, I appreciate the work you are doing.

The change Aaron made for indi allsky was to take into account AWB taking up to 5 images to find the correct white balance; this meant that a 100 second exposure could take up to 500 seconds to complete, and indi allsky would timeout after 300 seconds.  If @ldb20151 was using AWB that could explain his "hangs" which would not have been hangs, just very long running libcamera-still commands.  In that case, do you know what changed in libcamera between versions 0.3y and 0.5y that would impact how long libcamera-still takes?  Did the number of possible images in AWB increase?

It's likely the problem above is different than https://github.com/raspberrypi/rpicam-apps/issues/799 since that "hang" was a true hang per sdb.

ldb20151, 2025-07-12:
By the way I have tried running the simple image capture loop to see if this lockup will happen. My attempts so far have failed to recreate the issue; how ever I have must run the loop in 1 day increments. Due to my schedule I have not been able to just let it run. 

Since I’m planning to build a second system I could clone my first system and run a loop on it #2. I’ll do that when I get the  second camera.