raspberrypi/libcamera #278: 16bit mode with embedded data line and optical blank read as main frame data

will127534, 2025-06-26:
Hi there,

I'm working on IMX585 driver for upstreaming and thus start to look at the ClearHDR 16bit mode more closely.
I noticed that with 16bit mode and capture data with rpicam-raw, I'm seeing additional line and optical blank in the RPI5 captured data while RPI4/CM4 did not.

Here is the capture with RPI5, zoomed in to the top left corner
[image]

Here is the same driver captured from CM4, zoomed in to the top left corner
[image]

Not quite sure what info/debug log needed to do further debug so let me know if you need more info.

Thanks,
WillWhang

will127534, 2025-06-26:
The first row is actually embedded data, and the dark rows are optical blank.
Same mode on VC4 I didn't see those shows up in the raw data.

0x37 for Optical Blank, 0x12 for embedded data, 0x2E for framedata.
Should be the standard CSI-2 16bit frame format.

naushir, 2025-06-26:
This is possibly related to the discussion on https://github.com/raspberrypi/libcamera/issues/99 about reverting https://github.com/raspberrypi/linux/commit/98b1f0dbb23faf3e96005a25f6acb95b64afce43.  Is this commit reverted in your tree?

will127534, 2025-06-26:
Yes it is

naushir, 2025-06-26:
Ok, with the change in https://github.com/raspberrypi/linux/pull/6926, you ought to be able to get the image data in the main buffer, and if enabled, embedded data + optical shielded pixels in the second buffer.  You should not have to revert https://github.com/raspberrypi/linux/commit/98b1f0dbb23faf3e96005a25f6acb95b64afce43 either.

Note that as before, the statistics for the 16-bit format will not be usable.

will127534, 2025-06-26:
Oh wait I just realized that is a update 7 hours ago, will try it later

will127534, 2025-06-26:
Okkk with the latest 6.12.y, I'm still seeing the same thing

[image]

naushir, 2025-06-27:
https://github.com/raspberrypi/linux/pull/6926 has not been merged yet.  Presumably you built the kernel manually or got the change via rpi-update?

will127534, 2025-06-27:
Ah I see, yes I build/pull the kernel from the latest 6.12.y branch but not the change, I thought it is the latest one because I see you force-push something.

naushir, 2025-06-27:
You can get the kernel from CI with

```
sudo rpi-update pulls/6926
```

will127534, 2025-06-27:
Thanks! Got the patch up and running, but I don't think it is working.
Here is the patch:

[image]

Here is the preview from rpicam:
[image]
Here is the raw file
[image]

Guess I really need to have a test pattern to see what's going on, but I don't have the datasheet info on that at the moment.

naushir, 2025-06-27:
In your viewer, can you switch the endan order and see if that makes a difference?

will127534, 2025-06-27:
Yeap I did try that but the raw file did not show much difference

[image]
I have debayer disabled here so the blank lines are really weird.

naushir, 2025-06-27:
As a quick workaround, is it possible to set/change the packet id for image data from the sensor?

will127534, 2025-06-27:
I don't think there is a way to do that sadly...

will127534, 2025-06-27:
I wonder why reverting commit 98b1f0d we see the data correctly but with EB(embedded data) and OB (Optical Blank).
Will enabling CFE debug message helps here?

naushir, 2025-06-27:
> I wonder why reverting commit 98b1f0d we see the data correctly but with EB(embedded data) and OB (Optical Blank).

I can only speculate right now, but I think when the packet id capture is set to 0 (as when we revert the comit), the hardware disables auto-unpacking (which is good, because the auto-unpacking is wrong because of a HW bug).  So we end up with the semi-correct (endian is still wrong) RAW data.  But because we use a wildcard capture of 0, all other data packages (embedded + optical) also get routed into the buffer.

What we need is a neat solution where the auto unpacking is disabled, but we also route the embedded + optical data our of the image buffer.

naushir, 2025-06-27:
Yes, that is possible - but the key is how to implement something as cleanly as possible.  There's no option of disabling OB and embedded data in the sensor right?

will127534, 2025-06-27:
I don't think there is a way to do so on the sensor side

will127534, 2025-07-04:
Hi @naushir, I'm able to turn off embedded data line and it look fine now:

[image]
(Notice the OB black bar is still there but the RGGB pattern is correct).

But question for you is do we / can we revert that kernel commit? 

I'm seeing the following kernel message:
```
[   68.143332] ------------[ cut here ]------------
[   68.143338] WARNING: CPU: 3 PID: 835 at drivers/media/platform/raspberrypi/rp1_cfe/cfe.c:855 cfe_start_streaming+0x8c0/0x900 [rp1_cfe]
[   68.143348] Modules linked in: algif_hash algif_skcipher af_alg rp1_pio bnep binfmt_misc imx585(O) brcmfmac_wcc spidev hci_uart btbcm bluetooth vc4 aes_ce_blk brcmfmac aes_ce_cipher ghash_ce gf128mul sha2_ce sha256_arm64 snd_soc_hdmi_codec sha1_ce brcmutil drm_display_helper cec sha1_generic raspberrypi_hwmon drm_dma_helper cfg80211 snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm ecdh_generic rpivid_hevc(C) ecc libaes rp1_cfe pisp_be rfkill v4l2_mem2mem v4l2_fwnode v4l2_async videobuf2_dma_contig snd_timer videobuf2_memops videobuf2_v4l2 videodev snd v3d spi_bcm2835 videobuf2_common i2c_brcmstb gpio_keys gpu_sched mc drm_shmem_helper drm_kms_helper pwm_fan rp1_adc i2c_designware_platform rp1 i2c_designware_core raspberrypi_gpiomem rp1_mailbox nvmem_rmem uio_pdrv_genirq uio i2c_dev drm fuse drm_panel_orientation_quirks backlight dm_mod ip_tables x_tables ipv6
[   68.143414] CPU: 3 UID: 1000 PID: 835 Comm: rpicam-still Tainted: G        WC O       6.12.17-v8-16k+ #1
[   68.143418] Tainted: [W]=WARN, [C]=CRAP, [O]=OOT_MODULE
[   68.143419] Hardware name: Raspberry Pi 5 Model B Rev 1.1 (DT)
[   68.143421] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   68.143424] pc : cfe_start_streaming+0x8c0/0x900 [rp1_cfe]
[   68.143428] lr : cfe_start_streaming+0x478/0x900 [rp1_cfe]
[   68.143431] sp : ffffc00083b43ac0
[   68.143432] x29: ffffc00083b43ac0 x28: ffffd06f52950be8 x27: 0000000000000000
[   68.143436] x26: ffff800104459a80 x25: ffff800105814000 x24: ffff800105810000
[   68.143439] x23: ffff800105813b08 x22: ffff800105814000 x21: ffff800105813368
[   68.143442] x20: ffffd06f52950490 x19: ffff800105810000 x18: 0000000000000000
[   68.143446] x17: 0000000000000000 x16: ffffd06fcf198288 x15: 00007fff900316c0
[   68.143449] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[   68.143452] x11: 0000000000000000 x10: 0000000000000000 x9 : ffffd06f5293fb00
[   68.143456] x8 : ffffc00083b43da8 x7 : 0000000000000005 x6 : 0000000000000884
[   68.143459] x5 : 0000000000000f10 x4 : 0000000000003020 x3 : 0000000000003020
[   68.143462] x2 : 0000000000000018 x1 : 0000000000000000 x0 : ffff8001008f7800
[   68.143466] Call trace:
[   68.143467]  cfe_start_streaming+0x8c0/0x900 [rp1_cfe]
[   68.143471]  vb2_start_streaming+0x74/0x190 [videobuf2_common]
[   68.143478]  vb2_core_streamon+0x108/0x1d8 [videobuf2_common]
[   68.143483]  vb2_ioctl_streamon+0x54/0xa0 [videobuf2_v4l2]
[   68.143489]  v4l_streamon+0x2c/0x40 [videodev]
[   68.143509]  __video_do_ioctl+0x170/0x3e0 [videodev]
[   68.143525]  video_usercopy+0x1bc/0x780 [videodev]
[   68.143540]  video_ioctl2+0x20/0x38 [videodev]
[   68.143556]  v4l2_ioctl+0x48/0x70 [videodev]
[   68.143571]  __arm64_sys_ioctl+0xb4/0x100
[   68.143576]  invoke_syscall+0x50/0x120
[   68.143581]  el0_svc_common.constprop.0+0x48/0xf0
[   68.143585]  do_el0_svc+0x24/0x38
[   68.143588]  el0_svc+0x30/0xd0
[   68.143593]  el0t_64_sync_handler+0x100/0x130
[   68.143596]  el0t_64_sync+0x190/0x198
[   68.143599] ---[ end trace 0000000000000000 ]---

```

will127534, 2025-07-04:
we would also need to add 16bit mode to rpicam-apps
Specifically this line:
https://github.com/raspberrypi/rpicam-apps/blob/main/core/rpicam_app.cpp#L43

naushir, 2025-07-04:
@will127534 good to hear that using embedded data seems to work better.

> But question for you is do we / can we revert that kernel commit?

Unfortunately, at this point you will have to revert the commit, i.e. the DT must be set to 0.  @njhollinghurst what do you think about simply setting up our format table that way in the driver, since the 16-bit mode is fundamentally broken anyway? 

> I'm seeing the following kernel message:

This is expected, since the DT we are using is now set to 0.  I can supress the warning though.  Let me make a PR for that.

One thing to note, if you really wanted, you can crop the frame in the CFE to remove the optical pixels.  However, there will be quite a bit of plumbing required in the code to enable this.

naushir, 2025-07-04:
PR raised: https://github.com/raspberrypi/linux/pull/6935.

will127534, 2025-07-04:
Thank you so much for the help!
I think I'm good with having the OB in the frames as it isn't that much of an issue and can be post processed fairly easily.

will127534, 2025-07-13:
Hi @naushir, thanks a lot for the help!

Can you also help me adding 16bit mode in rpicam-apps?

naushir, 2025-07-14:
Should be available in https://github.com/raspberrypi/rpicam-apps/pull/836