raspberrypi/libcamera #283: Pipeline/PISP: libcamera::controls::FrameWallClock is returning time in microseconds

larsll, 2025-07-04:
When accessing the `libcamera::controls::FrameWallClock` on a Raspberry Pi5 using a Camera Module 3 the time returned is in microseconds, not nanoseconds.

Cause should be here:
https://github.com/raspberrypi/libcamera/blob/d83ff0a4ae4503bc56b7ed48cd142c3dd423ad3b/src/libcamera/pipeline/rpi/pisp/pisp.cpp#L1749

I do not see a reason why the sensorTimestamp is divided by 1000.

naushir, 2025-07-07:
Unless I'm mistaken, ``wallClockTimestamp`` is in units of us.  So since ``sensorTimestamp`` is in ns, we want to divide by 1000.

larsll, 2025-07-07:
Description says:
> This timestamp corresponds to the same moment in time as the
> SensorTimestamp, but is represented as a wall clock time as measured by
> the CLOCK_REALTIME clock.

SensorTimestamp returns the time in nanoseconds, clock_gettime(CLOCK_BOOTTIME, ...) is seconds + nanoseconds, so I don't see a documented reason why FrameWallClock should return microseconds.

naushir, 2025-07-07:
Well we introduced this control for our frame sync algorithm and we didn't  have any practical use for nanosecond precision for it.

However we are talking to the libcamera dev team about increasing the resolution to ns in the near future.

larsll, 2025-07-10:
> Well we introduced this control for our frame sync algorithm and we didn't have any practical use for nanosecond precision for it.
> 
> However we are talking to the libcamera dev team about increasing the resolution to ns in the near future.

Ok. Then I would suggest to add to the documentation that this is returned in us, as changing it later will then also be a potentially breaking change.

kbingham, 2025-07-10:
the next libcamera release upstream will be in about 3 weeks (I try to align with the kernel release schedule) so this shouldn't count as a breaking change ... (upstream) ... because it will be introduced and 'fixed' in the same release cycle.

closed 2025-08-10 by larsll