raspberrypi/libcamera #287: use multiple streams with isp

tranhien1612, 2025-07-26:
i read examples and saw ISP can support 3 streams running concurrently (Main, viewfinder, still capture). but i cant find any example about it. i want to use more than 1 stream for my camera applications. 

Specifically I want to run at least 2 stream, 1 for streaming and 1 for capturing images.

i don't know how to do it. can anyone help me do it? thanks for all.
```
	 *   +-------------------------------------------------------+
	 *   | Camera                                                |
	 *   |                +-----------+                          |
	 *   | +--------+     |           |------> [  Main output  ] |
	 *   | | Image  |     |           |                          |
	 *   | |        |---->|    ISP    |------> [   Viewfinder  ] |
	 *   | | Source |     |           |                          |
	 *   | +--------+     |           |------> [ Still Capture ] |
	 *   |                +-----------+                          |
	 *   +-------------------------------------------------------+
```

naushir, 2025-07-28:
Which Raspberry Pi platform are you running on? You can only generate 2x simultaneous output from the ISP.  If you have a look at some of our example demos, particularly the [Hailo postprocessing ](https://github.com/raspberrypi/rpicam-apps/tree/main/post_processing_stages/hailo) stages, you can see how to setup the hardware to do this.

Another much easier way would be to use Picamera2, for example, [this script](https://github.com/raspberrypi/picamera2/blob/main/examples/capture_video_multiple.py) captures 2 streams and encodes them independently.

naushir, 2025-07-28:
Have a look at the ``rpicam-apps`` example code I linked to above - that should provide examples of what you want in C++.

tranhien1612, 2025-07-30:
@naushir Hi, now i am creating a source code to do this. i use 2 config `StillCapture` and `Raw`. it seems it can run. But i don't know that my source code is right with my function, and i want to read image from 2 config in `requestComplete` function. and i don't know how to do it.

```

#include <condition_variable>
#include <iostream>
#include <memory>
#include <mutex>
#include <queue>
#include <set>
#include <string>
#include <thread>
#include <variant>
#include <any>
#include <map>
#include <iomanip>
#include <fstream>
#include <sys/mman.h>  // For mmap, munmap, PROT_READ, PROT_WRITE, MAP_SHARED
#include <fcntl.h> 

#include <libcamera/transform.h>
#include <libcamera/libcamera.h>
#include <opencv2/opencv.hpp>
#include <libcamera/base/span.h>
#include <libcamera/camera.h>
#include <libcamera/camera_manager.h>
#include <libcamera/control_ids.h>
#include <libcamera/controls.h>
#include <libcamera/formats.h>
#include <libcamera/framebuffer_allocator.h>
#include <libcamera/property_ids.h>

using namespace libcamera;
std::unique_ptr<CameraManager> manager;
std::shared_ptr<Camera> camera;

void requestComplete(Request *request)
{
	if (request->status() == Request::RequestCancelled)
		return;

    std::cout << std::endl << "Request completed: " << request->toString() << std::endl;
    
    const ControlList &requestMetadata = request->metadata();
	for (const auto &ctrl : requestMetadata) {
		const ControlId *id = controls::controls.at(ctrl.first);
		const ControlValue &value = ctrl.second;

		std::cout << "\t" << id->name() << " = " << value.toString() << std::endl;
	}

    const Request::BufferMap &buffers = request->buffers();
	for (auto bufferPair : buffers) {
		// (Unused) Stream *stream = bufferPair.first;
		FrameBuffer *buffer = bufferPair.second;
		const FrameMetadata &metadata = buffer->metadata();

		/* Print some information about the buffer which has completed. */
		std::cout << " seq: " << std::setw(6) << std::setfill('0') << metadata.sequence
			  << " timestamp: " << metadata.timestamp
			  << " bytesused: ";

		unsigned int nplane = 0;
		for (const FrameMetadata::Plane &plane : metadata.planes()){
			std::cout << plane.bytesused;
			if (++nplane < metadata.planes().size())
				std::cout << "/";
		}

		std::cout << std::endl;
	}

    request->reuse(Request::ReuseBuffers);
	camera->queueRequest(request);
}

int main() {    
    manager = std::make_unique<CameraManager>();
    manager->start();
    std::cout << "Camera manager started" << std::endl;

    if (manager->cameras().empty()) {
        std::cerr << "No cameras found!" << std::endl;
        return 1;
    }

    camera = manager->cameras()[0];
    std::cout << "Found camera: " << camera->id() << std::endl;

    if (camera->acquire()) {
        std::cerr << "Failed to acquire camera" << std::endl;
        return 1;
    }
    std::cout << "Camera acquired successfully" << std::endl;

    // Configure two streams
    std::unique_ptr<CameraConfiguration> configuration =
		camera->generateConfiguration( { StreamRole::StillCapture, StreamRole::Raw } );
    for (size_t i = 0; i < configuration->size(); ++i) {
        StreamConfiguration &config = configuration->at(i);
        std::cout << "Stream " << i << " configuration: " << config.toString() << std::endl;
    }

    // Validate configuration
    configuration->validate();

    if (camera->configure(configuration.get())) {
        std::cerr << "Configuration failed" << std::endl;
        return 1;
    }
    std::cout << "Camera configured successfully" << std::endl;

    // Allocator
    FrameBufferAllocator *allocator = new FrameBufferAllocator(camera);
    for (StreamConfiguration &config : *configuration){
        int ret = allocator->allocate(config.stream());
        if (ret < 0) {
            std::cerr << "Can't allocate buffers" << std::endl;
            return -1;
        }

        size_t allocated = allocator->buffers(config.stream()).size();
        std::cout << "Allocated " << allocated << " buffers for stream" << std::endl;
    }

    // Frame Capture
    std::vector<std::unique_ptr<Request>> requests;

    Stream *stream0 = configuration->at(0).stream();
    Stream *stream1 = configuration->at(1).stream();

    const std::vector<std::unique_ptr<FrameBuffer>> &buffers0 = allocator->buffers(stream0);
    const std::vector<std::unique_ptr<FrameBuffer>> &buffers1 = allocator->buffers(stream1);

    for (unsigned int i = 0; i < buffers0.size(); ++i) {
		std::unique_ptr<Request> request = camera->createRequest();
		if (!request){
			std::cerr << "Can't create request" << std::endl;
			return EXIT_FAILURE;
		}

		const std::unique_ptr<FrameBuffer> &buffer = buffers0[i];
		int ret = request->addBuffer(stream0, buffer.get());
		if (ret < 0){
			std::cerr << "Can't set buffer for request" << std::endl;
			return EXIT_FAILURE;
		}
		requests.push_back(std::move(request));
	}

    for (unsigned int i = 0; i < buffers1.size(); ++i) {
		std::unique_ptr<Request> request = camera->createRequest();
		if (!request){
			std::cerr << "Can't create request" << std::endl;
			return EXIT_FAILURE;
		}

		const std::unique_ptr<FrameBuffer> &buffer = buffers1[i];
		int ret = request->addBuffer(stream1, buffer.get());
		if (ret < 0){
			std::cerr << "Can't set buffer for request" << std::endl;
			return EXIT_FAILURE;
		}
		requests.push_back(std::move(request));
	}

    // Signal & Slot
    camera->requestCompleted.connect(requestComplete);

    // Start Capture
    camera->start();
    for (std::unique_ptr<Request> &request : requests)
		camera->queueRequest(request.get());

    while(true);

    camera->stop();
	delete allocator;
	allocator = nullptr;
    camera->release();
    camera.reset();
    manager->stop();

    return 0;
}
```

naushir, 2025-07-30:
```
const Request::BufferMap &buffers = request->buffers();
```

This will return a buffer map object ``using BufferMap = std::map<const Stream *, FrameBuffer *>;``, so if you enumerate through this, you will have one ``FrameBuffer *`` pointer for each ``Stream`` object that you have requested.

naushir, 2025-07-30:
Which sensor are you using in your platform?

tranhien1612, 2025-07-30:
> Which sensor are you using in your platform?

i used `Raspberry Pi Camera Module V3` in `CM4 dev kit`.

naushir, 2025-07-30:
Might I suggest you look at ``rpicam-apps`` (https://github.com/raspberrypi/rpicam-apps/), which does exactly this, i.e. can capture multiple streams (RAW and/or YUV)?

tranhien1612, 2025-07-30:
i used picamera2 python, and it is ok for my response. but currently, i only use c++ for my request (because i must add more another function for my project, and it only run with c++)

tranhien1612, 2025-07-30:
I think it's because opencv's image saving process takes longer than `requestComplete`, which causes the image to be incompletely written. It seems like the recording process is taking quite a while.

naushir, 2025-07-30:
You should do the minimal amount of work in the requestComplete callback.  Ideally, you have a worker thread to do all the image processing and saving.

tranhien1612, 2025-07-30:
i used  the `queue ` to save request and handle it in thread. 
```
void requestComplete(Request *request){
	if (request->status() == Request::RequestCancelled)
		return;
    requestQueue.push(request);
}

void imageHandle(){
    std::lock_guard<std::mutex> lock(free_requests_mutex_);
    while(1){
        if (!requestQueue.empty()){
            Request *request = requestQueue.front();
            const Request::BufferMap &buffers = request->buffers();
            for (auto &[stream, buffer] : buffers) {
                int width = stream->configuration().size.width;
                int height = stream->configuration().size.height;
                int stride = stream->configuration().stride;

                if (stream == streamView) {
                    std::cout << "View Stream: " << width << ", " << height << ", tride: " << stride << ", sizeBuffer: " << buffer->planes().size() <<std::endl;
                }else if (stream == streamVideo) {
                    std::cout << "Video Stream: " << width << ", " << height << ", tride: " << stride << ", sizeBuffer: " << buffer->planes().size() <<std::endl;
                }else if (stream == streamStill) {
                    const FrameBuffer::Plane &plane = buffer->planes()[0];
                    void *data = mappedBuffersStill_[plane.fd.get()].first;
                    if (!data) {
                        std::cerr << "Null data pointer for still stream!" << std::endl;
                        continue;
                    }
                    cv::Mat image(height, width, CV_8UC3, data, stride);
                    cv::Mat copied = image.clone();
                    cv::imwrite("RGB888.png", copied);
                }else if (stream == streamRaw) {
                    const FrameBuffer::Plane &plane = buffer->planes()[0];
                    void *data = mappedBuffersRaw_[plane.fd.get()].first;
                    const FrameMetadata::Plane &meta = buffer->metadata().planes()[0];
                    int length = std::min(meta.bytesused, plane.length);
                    cv::Mat raw16(height, stride / 2, CV_16UC1, data); // divide stride because it's in bytes
                    cv::Mat cropped = raw16(cv::Rect(0, 0, width, height));
                    cv::Mat bayer10, bayer8, rgb;
                    cv::bitwise_and(cropped, 0x03FF, bayer10); // apply 10-bit mask
                    bayer10.convertTo(bayer8, CV_8UC1, 255.0 / 1023.0); // scale 10-bit to 8-bit
                    cv::cvtColor(bayer8, rgb, cv::COLOR_BayerBG2RGB); // for SBGGR10 use BG2RGB
                    cv::imwrite("SBGGR10.png", rgb);
                }
            }

            requestQueue.pop();
        }
    }
}
```
with image is created by still stream, it seems like ok, but image from raw stream with full resolution, it still has the same error.

[image]

and I don't know why it is like that.

naushir, 2025-07-30:
Can you try saving the RAW data directly (without any conversion) and inspect it with a tool like [PixelViewer](https://github.com/carina-studio/PixelViewer) please?  I'm not familiar with the OpenCV image libraries, so won't be able to help you with that.

naushir, 2025-07-30:
The RAW stream consists of Bayer samples directly from the sensor, i.e. entirely unprocessed.  It will look much darker than the YUV/RGB streams as they have full ISP processing (colour gains, colour correction, gamma, etc.) applied to the samples.

naushir, 2025-08-04:
Any more help needed on this?  If not, I'll close this shortly.

tranhien1612, 2025-08-04:
@naushir Thank for you suggettion, i got it