raspberrypi/libcamera #298: Unable to acquire a CFE instance

jwinarske, 2025-08-22:
I'm working on adding camera support to a Yocto build.  It fails with `Unable to acquire a CFE instance`.

* libcamera 0.5.1
* libpisp 1.2.1

```
root@raspberrypi5:~# dmesg |grep imx
[    0.029628] /axi/pcie@1000120000/rp1/i2c@80000/imx708@1a: Fixed dependency cycle(s) with /axi/pcie@1000120000/rp1/csi@128000
[    0.029670] /axi/pcie@1000120000/rp1/csi@128000: Fixed dependency cycle(s) with /axi/pcie@1000120000/rp1/i2c@80000/imx708@1a
[    0.029857] /axi/pcie@1000120000/rp1/i2c@80000/imx708@1a: Fixed dependency cycle(s) with /axi/pcie@1000120000/rp1/csi@128000
[    0.029900] /axi/pcie@1000120000/rp1/csi@128000: Fixed dependency cycle(s) with /axi/pcie@1000120000/rp1/i2c@80000/imx708@1a
[    0.503023] /axi/pcie@1000120000/rp1/i2c@80000/imx708@1a: Fixed dependency cycle(s) with /axi/pcie@1000120000/rp1/csi@128000
[    0.516021] /axi/pcie@1000120000/rp1/i2c@80000/imx708@1a: Fixed dependency cycle(s) with /axi/pcie@1000120000/rp1/csi@128000
[    0.516036] /axi/pcie@1000120000/rp1/csi@128000: Fixed dependency cycle(s) with /axi/pcie@1000120000/rp1/i2c@80000/imx708@1a
[    2.198355] rp1-cfe 1f00128000.csi: found subdevice /axi/pcie@1000120000/rp1/i2c@80000/imx708@1a
[    2.400630] imx708 11-001a: camera module ID 0x0382
[    2.403745] rp1-cfe 1f00128000.csi: Using sensor imx708_wide_noir for capture
[    2.644597] imx219 11-0010: Error reading reg 0x0000: -121
[    2.650124] imx219 11-0010: error -EREMOTEIO: failed to read chip id 219
[    2.660007] imx219 11-0010: probe with driver imx219 failed with error -121
root@raspberrypi5:~# dmesg |grep rp1-cfe
[    2.219561] rp1-cfe 1f00128000.csi: bcm2712_iommu_of_xlate: MMU 1000005280.iommu
[    2.219571] rp1-cfe 1f00128000.csi: bcm2712_iommu_probe_device: MMU 1000005280.iommu
[    2.219577] rp1-cfe 1f00128000.csi: bcm2712_iommu_device_group: MMU 1000005280.iommu
[    2.219582] rp1-cfe 1f00128000.csi: Adding to iommu group 2
[    2.219586] rp1-cfe 1f00128000.csi: bcm2712_iommu_attach_dev: MMU 1000005280.iommu
[    2.220006] rp1-cfe 1f00128000.csi: DW dphy Host HW v1.20
[    2.220026] rp1-cfe 1f00128000.csi: PiSP FE HW v0.1
[    2.221323] rp1-cfe 1f00128000.csi: found subdevice /axi/pcie@1000120000/rp1/i2c@80000/imx708@1a
[    2.362421] rp1-cfe 1f00128000.csi: Using sensor imx708_wide_noir for capture
[    2.362496] rp1-cfe 1f00128000.csi: Registered [rp1-cfe-csi2_ch0] node id 0 successfully as /dev/video0
[    2.362553] rp1-cfe 1f00128000.csi: Registered [rp1-cfe-embedded] node id 1 successfully as /dev/video1
[    2.362586] rp1-cfe 1f00128000.csi: Registered [rp1-cfe-csi2_ch2] node id 2 successfully as /dev/video2
[    2.363884] rp1-cfe 1f00128000.csi: Registered [rp1-cfe-csi2_ch3] node id 3 successfully as /dev/video3
[    2.364098] rp1-cfe 1f00128000.csi: Registered [rp1-cfe-fe_image0] node id 4 successfully as /dev/video4
[    2.364151] rp1-cfe 1f00128000.csi: Registered [rp1-cfe-fe_image1] node id 5 successfully as /dev/video5
[    2.369164] rp1-cfe 1f00128000.csi: Registered [rp1-cfe-fe_stats] node id 6 successfully as /dev/video6
[    2.369498] rp1-cfe 1f00128000.csi: Registered [rp1-cfe-fe_config] node id 7 successfully as /dev/video7
root@raspberrypi5:~# lsmod | grep rp1_cfe
rp1_cfe                65536  0
v4l2_fwnode            49152  4 imx219,imx708,rp1_cfe
v4l2_async             49152  5 v4l2_fwnode,imx219,imx708,rp1_cfe,dw9807_vcm
videobuf2_dma_contig    49152  3 pisp_be,rpi_hevc_dec,rp1_cfe
videobuf2_v4l2         49152  4 pisp_be,rpi_hevc_dec,rp1_cfe,v4l2_mem2mem
videodev              344064  10 v4l2_async,v4l2_fwnode,imx219,pisp_be,rpi_hevc_dec,imx708,videobuf2_v4l2,rp1_cfe,dw9807_vcm,v4l2_mem2mem
videobuf2_common       98304  7 pisp_be,rpi_hevc_dec,videobuf2_dma_contig,videobuf2_v4l2,rp1_cfe,v4l2_mem2mem,videobuf2_memops
mc                     81920  11 v4l2_async,videodev,imx219,pisp_be,rpi_hevc_dec,imx708,videobuf2_v4l2,rp1_cfe,videobuf2_common,dw9807_vcm,v4l2_mem2mem
root@raspberrypi5:~# v4l2-ctl --all
Driver Info:
	Driver name      : rp1-cfe
	Card type        : rp1-cfe
	Bus info         : platform:1f00128000.csi
	Driver version   : 6.12.41
	Capabilities     : 0xaca00001
		Video Capture
		Metadata Capture
		Metadata Output
		I/O MC
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x24a00001
		Video Capture
		Metadata Capture
		I/O MC
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : rp1-cfe
	Model            : rp1-cfe
	Serial           : 
	Bus info         : platform:1f00128000.csi
	Media version    : 6.12.41
	Hardware revision: 0x00114666 (1132134)
	Driver version   : 6.12.41
Interface Info:
	ID               : 0x03000017
	Type             : V4L Video
Entity Info:
	ID               : 0x00000015 (21)
	Name             : rp1-cfe-csi2_ch0
	Function         : V4L2 I/O
	Pad 0x01000016   : 0: Sink, Must Connect
	  Link 0x02000037: from remote pad 0x1000006 of entity 'csi2' (Video Interface Bridge): Data
Priority: 2
Video input : 0 (rp1-cfe-csi2_ch0: ok)
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'pRAA' (10-bit Bayer RGRG/GBGB Packed)
	Field             : None
	Bytes per Line    : 800
	Size Image        : 384000
	Colorspace        : Raw
	Transfer Function : None
	YCbCr/HSV Encoding: ITU-R 601
	Quantization      : Full Range
	Flags             : 
Format Metadata Capture:
	Sample Format   : 'SENS' (Sensor Ancillary Metadata)
	Buffer Size     : 16384
root@raspberrypi5:~# LIBCAMERA_LOG_LEVELS=*:DEBUG cam -l
[0:01:57.181524299] [515] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_pisp.so: IPA module /usr/lib/libcamera/ipa/ipa_rpi_pisp.so is signed
[0:01:57.181602280] [515] DEBUG IPAManager ipa_manager.cpp:229 Loaded IPA module '/usr/lib/libcamera/ipa/ipa_rpi_pisp.so'
[0:01:57.181618743] [515]  INFO Camera camera_manager.cpp:326 libcamera v0.5.1+dirty (2025-08-22T20:34:07UTC)
[0:01:57.181746651] [516] DEBUG Camera camera_manager.cpp:73 Starting camera manager
[0:01:57.191473306] [516] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media2
[0:01:57.191724769] [516] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 10 missing dependencies
[0:01:57.193156085] [516] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[0:01:57.193169029] [516] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: rp1-cfe
[0:01:57.193256437] [516] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpi-hevc-dec" created from /dev/media3
[0:01:57.193264641] [516] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 1 missing dependencies
[0:01:57.193408196] [516] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[0:01:57.193415067] [516] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: rpi-hevc-dec
[0:01:57.193507326] [516] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media0
[0:01:57.193516141] [516] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 8 missing dependencies
[0:01:57.193614956] [516] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media1
[0:01:57.193623567] [516] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 8 missing dependencies
[0:01:57.194687604] [516] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[0:01:57.194695216] [516] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: pispbe
[0:01:57.195761827] [516] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[0:01:57.195769827] [516] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: pispbe
[0:01:57.196042754] [516] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'rpi/pisp'
[0:01:57.197212791] [516] DEBUG RPI pisp.cpp:890 Unable to acquire a CFE instance
Available cameras:
root@raspberrypi5:~# 
```

What am I missing?

jwinarske, 2025-08-22:
Same result with a single sensor enabled/attached.
```
root@raspberrypi5:~# dmesg |grep imx
[    0.029518] /axi/pcie@1000120000/rp1/i2c@80000/imx477@1a: Fixed dependency cycle(s) with /axi/pcie@1000120000/rp1/csi@128000
[    0.029559] /axi/pcie@1000120000/rp1/csi@128000: Fixed dependency cycle(s) with /axi/pcie@1000120000/rp1/i2c@80000/imx477@1a
[    0.029741] /axi/pcie@1000120000/rp1/i2c@80000/imx477@1a: Fixed dependency cycle(s) with /axi/pcie@1000120000/rp1/csi@128000
[    0.029783] /axi/pcie@1000120000/rp1/csi@128000: Fixed dependency cycle(s) with /axi/pcie@1000120000/rp1/i2c@80000/imx477@1a
[    0.502929] /axi/pcie@1000120000/rp1/i2c@80000/imx477@1a: Fixed dependency cycle(s) with /axi/pcie@1000120000/rp1/csi@128000
[    0.515922] /axi/pcie@1000120000/rp1/i2c@80000/imx477@1a: Fixed dependency cycle(s) with /axi/pcie@1000120000/rp1/csi@128000
[    0.515937] /axi/pcie@1000120000/rp1/csi@128000: Fixed dependency cycle(s) with /axi/pcie@1000120000/rp1/i2c@80000/imx477@1a
[    2.204227] rp1-cfe 1f00128000.csi: found subdevice /axi/pcie@1000120000/rp1/i2c@80000/imx477@1a
[    2.350481] imx477 11-001a: Device found is imx477
[    2.350526] rp1-cfe 1f00128000.csi: Using sensor imx477 11-001a for capture
root@raspberrypi5:~# LIBCAMERA_LOG_LEVELS=*:DEBUG cam -l
[0:04:22.279223568] [507] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_pisp.so: IPA module /usr/lib/libcamera/ipa/ipa_rpi_pisp.so is signed
[0:04:22.279310957] [507] DEBUG IPAManager ipa_manager.cpp:229 Loaded IPA module '/usr/lib/libcamera/ipa/ipa_rpi_pisp.so'
[0:04:22.279331198] [507]  INFO Camera camera_manager.cpp:326 libcamera v0.5.1+dirty (2025-08-22T20:34:07UTC)
[0:04:22.279452828] [508] DEBUG Camera camera_manager.cpp:73 Starting camera manager
[0:04:22.288576182] [508] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media0
[0:04:22.288714404] [508] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 10 missing dependencies
[0:04:22.290112222] [508] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[0:04:22.290130704] [508] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: rp1-cfe
[0:04:22.290221223] [508] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpi-hevc-dec" created from /dev/media3
[0:04:22.290238112] [508] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 1 missing dependencies
[0:04:22.290391186] [508] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[0:04:22.290406742] [508] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: rpi-hevc-dec
[0:04:22.290511612] [508] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media1
[0:04:22.290529297] [508] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 8 missing dependencies
[0:04:22.290637501] [508] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media2
[0:04:22.290655261] [508] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 8 missing dependencies
[0:04:22.291726467] [508] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[0:04:22.291743559] [508] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: pispbe
[0:04:22.292819340] [508] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[0:04:22.292837136] [508] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: pispbe
[0:04:22.293101655] [508] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'rpi/pisp'
[0:04:22.293133433] [508] DEBUG RPI pisp.cpp:890 Unable to acquire a CFE instance
Available cameras:
root@raspberrypi5:~#
```

jwinarske, 2025-08-25:
@kbingham If I use tip of tree from `next` it just works.  Log below.

Outstanding questions (as to correctly model it in Yocto support):
* Which branch should be used with what?
* How to determine logical release version for next?
* How to determine version association with libpisp?

```
root@raspberrypi5:/home/weston# LIBCAMERA_LOG_LEVELS=*:DEBUG cam -l
[65:54:52.933525947] [5319] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_pisp.so: IPA module /usr/lib/libcamera/ipa/ipa_rpi_pisp.so is signed
[65:54:52.933592651] [5319] DEBUG IPAManager ipa_manager.cpp:229 Loaded IPA module '/usr/lib/libcamera/ipa/ipa_rpi_pisp.so'
[65:54:52.933673633] [5319] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_vc4.so: IPA module /usr/lib/libcamera/ipa/ipa_rpi_vc4.so is signed
[65:54:52.933691189] [5319] DEBUG IPAManager ipa_manager.cpp:229 Loaded IPA module '/usr/lib/libcamera/ipa/ipa_rpi_vc4.so'
[65:54:52.933698744] [5319]  INFO Camera camera_manager.cpp:330 libcamera v0.5.1+150-c576b437-dirty (2025-08-25T17:26:32UTC)
[65:54:52.933828486] [5320] DEBUG Camera camera_manager.cpp:73 Starting camera manager
[65:54:52.943060749] [5320] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media0
[65:54:52.943196768] [5320] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 10 missing dependencies
[65:54:52.944609352] [5320] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[65:54:52.944621611] [5320] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: rp1-cfe
[65:54:52.944706408] [5320] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpi-hevc-dec" created from /dev/media1
[65:54:52.944716167] [5320] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 1 missing dependencies
[65:54:52.944864483] [5320] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[65:54:52.944872094] [5320] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: rpi-hevc-dec
[65:54:52.944966113] [5320] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media2
[65:54:52.944975298] [5320] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 8 missing dependencies
[65:54:52.945075632] [5320] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media3
[65:54:52.945084707] [5320] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 8 missing dependencies
[65:54:52.946151176] [5320] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[65:54:52.946160639] [5320] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: pispbe
[65:54:52.947216554] [5320] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[65:54:52.947223739] [5320] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: pispbe
[65:54:52.947479685] [5320] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'rpi/pisp'
[65:54:52.947508852] [5320] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "rp1-cfe"
[65:54:52.947522593] [5320] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "pispbe"
[65:54:52.947596019] [5320] DEBUG ClockRec clock_recovery.cpp:69 configure 100 2000 10 50000
[65:54:52.947602834] [5320] DEBUG ClockRec clock_recovery.cpp:83 reset
[65:54:52.947659168] [5320]  INFO RPI pisp.cpp:720 libpisp version v1.2.1 981977ff21f3-dirty 22-08-2025 (20:30:15)
[65:54:52.948286283] [5320] DEBUG CameraSensor camera_sensor_raw.cpp:210 imx477 11-001a: unsupported number of sinks (0) or sources (2)
[65:54:52.948334857] [5320] DEBUG V4L2 v4l2_device.cpp:721 'imx477 11-001a': Control: Exposure (0x00980911)
[65:54:52.948359654] [5320] DEBUG V4L2 v4l2_device.cpp:721 'imx477 11-001a': Control: Horizontal Flip (0x00980914)
[65:54:52.948376228] [5320] DEBUG V4L2 v4l2_device.cpp:721 'imx477 11-001a': Control: Vertical Flip (0x00980915)
[65:54:52.948393746] [5320] DEBUG V4L2 v4l2_device.cpp:721 'imx477 11-001a': Control: Camera Orientation (0x009a0922)
[65:54:52.948415654] [5320] DEBUG V4L2 v4l2_device.cpp:721 'imx477 11-001a': Control: Camera Sensor Rotation (0x009a0923)
[65:54:52.948432117] [5320] DEBUG V4L2 v4l2_device.cpp:721 'imx477 11-001a': Control: Vertical Blanking (0x009e0901)
[65:54:52.948447321] [5320] DEBUG V4L2 v4l2_device.cpp:721 'imx477 11-001a': Control: Horizontal Blanking (0x009e0902)
[65:54:52.948461432] [5320] DEBUG V4L2 v4l2_device.cpp:721 'imx477 11-001a': Control: Analogue Gain (0x009e0903)
[65:54:52.948476210] [5320] DEBUG V4L2 v4l2_device.cpp:721 'imx477 11-001a': Control: Red Pixel Value (0x009e0904)
[65:54:52.948491414] [5320] DEBUG V4L2 v4l2_device.cpp:721 'imx477 11-001a': Control: Green (Red) Pixel Value (0x009e0905)
[65:54:52.948506766] [5320] DEBUG V4L2 v4l2_device.cpp:721 'imx477 11-001a': Control: Blue Pixel Value (0x009e0906)
[65:54:52.948521673] [5320] DEBUG V4L2 v4l2_device.cpp:721 'imx477 11-001a': Control: Green (Blue) Pixel Value (0x009e0907)
[65:54:52.948537470] [5320] DEBUG V4L2 v4l2_device.cpp:721 'imx477 11-001a': Control: Link Frequency (0x009f0901)
[... cut ...]
[65:54:52.961537739] [5320] DEBUG Camera camera_manager.cpp:161 Pipeline handler "rpi/pisp" matched
[65:54:52.961554239] [5320] DEBUG RPI pisp.cpp:890 Unable to acquire a CFE instance
[65:54:52.961567109] [5320] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'rpi/vc4'
[65:54:52.961580221] [5320] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
[65:54:52.961591535] [5320] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
Available cameras:
1: 'imx477' (/base/axi/pcie@1000120000/rp1/i2c@80000/imx477@1a)
```

kbingham, 2025-08-26:
@jwinarske Do you mean you've tested https://github.com/raspberrypi/libcamera/commits/main/ ... and that doesn't work ?

I can imagine it's easy to get 'mainline libcamera' (which head of tree is a branch called 'master') with raspberrypi/main

jwinarske, 2025-08-26:
@kbingham my clone sequence happens to be in a file path with the word master in it.  Reading further, the sequence checks the branch which is indeed main.  The default branch for the repo is main.  The main branch does not work with rpi5, as the clone sequence noted above shows.  This is the root problem of this issue.

jwinarske, 2025-08-26:
@naushir I see.  You use a different label pattern from upstream, but do not remove upstream tags.  Upstream tags are associated with master only; so when I checkout v0.5.1 it's master (upstream).  To get tip of tree I should refer to the release tags, which would be `v0.5.1+rpt20250722`

Thanks for the clarification!