raspberrypi/libcamera #307: About libcamera0.4 support issues

long568, 2025-09-19:
I developed a program on a Debian virtual machine that depends on libcamera 0.4. I manually compiled and installed libcamera 0.4 in the Debian virtual machine, and the included libcamera-ipa version is 0.4.
When I copied the program to a Raspberry Pi and ran it, I received an error message. After checking, I found that the version of libcamera-ipa, which libcamera 0.4 depends on on the Raspberry Pi, is 0.5.2. I tried to install libcamera-ipa version 0.4 on the Raspberry Pi using apt, but it wasn't listed as available for installation.
This prevented me from completing the development process in the virtual machine. So, is it possible to include libcamera-ipa version 0.4 in the apt repository for the Raspberry Pi to facilitate development?

kbingham, 2025-09-19:
Can you rebuild your application against libcamera-0.5?

- when the libcamera 0.X version number changes that signifies that they are not compatible (I know this would usually be the 'major' number, but as this is a 0 ... it's bumped down to the next digit).

So an application built with libcamera-0.4 can not work (it's not abi-compatible) with libcamera-0.5.

kbingham, 2025-09-21:
You need to have the same versions of each component. Do not mix and match 0.4 with 0.5.2.

long568, 2025-09-21:
> You need to have the same versions of each component. Do not mix and match 0.4 with 0.5.2.

Got it.
But when I install libcamera 0.4 on my Raspberry Pi using apt , libcamera 0.5 and libcamera-ipa (0.5.2) are automatically installed.

kbingham, 2025-09-21:
Are you /explicitly/ installing libcamera-0.4 ? If so - why ? (And how?) Are you running 'apt install libcamera' or something more specific?

It could be that there’s something in Raspberry Pi's packaging relationships ... but ... don't stay on libcamera-0.4 if Raspberry Pi are shipping libcamera 0.5.2

Thinking about it - I doubt the IPA modules are built in a path that can support multiple versions to be installed at the same time. Maybe that's something that could be added as a feature if someone wants to work on it but it would be better to understand /why/ you are mixing versions in the first place.

naushir, 2025-09-22:
As @kbingham mentioned, there is no supported way to mix and match between libcamera base versions and IPA versions.  They *must* be the same.

naushir, 2025-10-01:
Closing this down due to inactivity. Please reopen if you still have problems.

closed 2025-10-01 by naushir