raspberrypi/libcamera #309: The request queue is empty forever

AnhPC03, 2025-09-22:
Hello @naushir 
My application calls to function `readFrame` in a while loop, after about 5 or 10 minutes, the `requestQueue` is always empty and the application never got image from camera again. Could you help me in this situation? Thank you very much.

```c++
#include "modules/utils/rpi_camera.h"

int RpiCamera::initCamera() {
    int ret;
    cm = std::make_unique<libcamera::CameraManager>();
    ret = cm->start();
    if (ret) {
        std::cout << "Failed to start camera manager: " << ret << std::endl;
        return ret;
    }
    cameraId = cm->cameras()[0]->id();
    camera_ = cm->get(cameraId);
    if (!camera_) {
        std::cerr << "Camera " << cameraId << " not found" << std::endl;
        return 1;
    }

    if (camera_->acquire()) {
        std::cerr << "Failed to acquire camera " << cameraId << std::endl;
        return 1;
    }
    camera_acquired_ = true;
    return 0;
}

char *RpiCamera::getCameraId() { return cameraId.data(); }

void RpiCamera::configureStill(int width, int height,
                               libcamera::PixelFormat format, int buffercount,
                               int rotation) {
    printf("Configuring still capture...\n");
    config_ =
        camera_->generateConfiguration({libcamera::StreamRole::StillCapture});
    if (width && height) {
        libcamera::Size size(width, height);
        config_->at(0).size = size;
    }
    config_->at(0).pixelFormat = format;
    if (buffercount) config_->at(0).bufferCount = buffercount;
    // libcamera::Transform transform = libcamera::Transform::Identity;
    // bool ok;
    // libcamera::Transform rot = libcamera::transformFromRotation(rotation,
    // &ok); if (!ok)
    //     throw std::runtime_error("illegal rotation value, Please use 0 or
    //     180");
    // transform = rot * transform;
    // if (!!(transform & libcamera::Transform::Transpose))
    //     throw std::runtime_error("transforms requiring transpose not
    //     supported");
    // config_->transform = transform;

    libcamera::CameraConfiguration::Status validation = config_->validate();
    if (validation == libcamera::CameraConfiguration::Invalid)
        throw std::runtime_error("failed to valid stream configurations");
    else if (validation == libcamera::CameraConfiguration::Adjusted)
        std::cout << "libcamera::Stream configuration adjusted" << std::endl;

    printf("Still capture setup complete\n");
}

int RpiCamera::startCamera() {
    int ret;
    ret = camera_->configure(config_.get());
    if (ret < 0) {
        std::cout << "Failed to configure camera" << std::endl;
        return ret;
[... 173 lines cut ...]
    while (!requestQueue.empty()) requestQueue.pop();

    for (auto &iter : mappedBuffers_) {
        std::pair<void *, unsigned int> pair_ = iter.second;
        munmap(std::get<0>(pair_), std::get<1>(pair_));
    }

    mappedBuffers_.clear();

    requests_.clear();

    allocator_.reset();

    // controls_.clear();
}

void RpiCamera::closeCamera() {
    if (camera_acquired_) camera_->release();
    camera_acquired_ = false;

    camera_.reset();

    cm.reset();
}
```

naushir, 2025-09-22:
You should check that ``RpiCamera::returnFrameBuffer()`` is called for every request.  If you don't recycle requests back into libcamera, things will stall like your case.  There's not enough code above to show how this function gets called.

AnhPC03, 2025-09-22:
@naushir thank you for your response
I am calling `returnFrameBuffer()` every request but got that error
```cpp
bool flag;
            RpiCameraOutData frameData;
            rpi_camera.startCamera();
            uint32_t stride;
            rpi_camera.VideoStream(&width, &height, &stride);
            int frame_count = 1;
            while (true) {
                if (terminate_program) {
                    break;
                }
                flag = rpi_camera.readFrame(&frameData);
                if (!flag) {
                    continue;
                }
                cv::Mat ori_frame(height, width, CV_8UC3, frameData.imageData,
                              stride);
                cv::Mat frame;
                cv::flip(ori_frame, frame, 0);
                ++frame_count;
                rpi_camera.returnFrameBuffer(frameData);
            }
            rpi_camera.stopCamera();
            rpi_camera.closeCamera();
```

naushir, 2025-09-22:
My only other suggestion would be to remove all external post-processing and try out a minimal application example .  Once that is working correctly, add back more functionality step by step.

naushir, 2025-09-22:
Yes that would be a start.